omersiar/esp-rfid #343: Opening the door by MQTT Command

Sansibari, 2019-12-10:
Change Request:
Remotly open the door via MQTT command and loging the event in the log file.

sinclairfr, 2019-12-30:
I concur, that would be GREAT !

ingeninge, 2020-01-02:
I tested with version 1.3.3 and it works. mosquitto as mqtt broker and node-red/dashboard (or mosquitto_pub on console).
You must not send a retained message, else you will have the door activated each time the esp-rfid reboots.
`mosquitto_pub -h IP.of.your.MQTTbroker -t rfid -m '{"cmd": "opendoor", "doorip": "192.168.xyz.xyz"}'
` where doorip is the ip address of the esp-rfid
The message is in JSON format. I hope it is readable inside the code segment.

wizznet1, 2020-01-22:
Hy all. Im new at esp rfid. I flash my esp with this new version. I managed to connect esp with cloudmqtt...but I dont know how to send messages to add user, delete user, activate relay 1,2,3,4....get door status and if there is some other commands. Please can someon send me commands and tell me how to do it?

frenchie71, 2020-01-26:
Hi @wizznet1 , as a starter have a look at the code in mqtt.esp of the dev branch, lines 176 following. The procedure onMqttMessage deals with incoming MQTT messages. 

First of all, you need to make sure you have the right topic. This is the base topic of your esp-rfid (i.e. the one you can listen to, the one you define in the setup page).

Next, you send a message under this topic.   The message needs to be in JSON format, i.e. something like  @ingeninge describes two posts above.

The command needs to be in the "cmd" key of that json string (lines 219++ of the code) - esp-rfid understands "listusr" , "opendoor", "deleteusers", "adduser".

depending on the command, additional keys need to be added. (For "adduser" you need to add the username in the "user" node and the uid (i.e. the PIN) in the "uid" node)

So if you wanted to add user "Jim" with Pincode "1234" to the esp-rfid with the IP address 10.10.10.10 you would send the following message under the esp-rfids topic:

`{"cmd": "adduser", "doorip": "10.10.10.10", "user": "Jim", "uid": "1234"}`

Hope this helps.

wizznet1, 2020-01-26:
Hy I will check it...do you know how to do it in mqtt dashboard app?

MHOMEASS, 2020-08-20:
I was new to this and it cost me a lot of time to figure out how to open the door from HomeAssitant. For those who are new to this here are the steps:

- use the latest firmware version (starting 1.3.1)
- go to HomeAssitant en go to 'developer tools'
- go to tab 'Services'
- select as service 'mqtt.publish'
- Service data use this data and fill in IP from the RFID (see status tab for IP) 
topic: /rfid
payload_template: '{"cmd": "opendoor", "doorip": "xxx.xxx.x.xxx"} '

If this works you can use the code in automations

matjack1, 2022-04-07:
this is present and documented in the `dev` branch, I'm closing this, but feel free to reopen if necessary!

closed 2022-04-07 by matjack1