omersiar/esp-rfid #344: Module relay 5v not working

adi160885, 2019-12-11:
sir, how to use the 5v relay module, because I use NodeMcu v3 there is no 5v output power only 3.3v, so the 5v relay module doesn't work, I've tried using (step up module) from 3.3v output but the 5v relay module doesn't work, please the solution, thank you.

kacitran, 2019-12-15:
----------relay signal
gpio -3.3v--------4.7k--------2n2222-|
                                                            -> ------ gnd

link photo
https://encrypted-tbn0.gstatic.com/images?q=tbn%3AANd9GcTyPjw2TzDluevxNvkuyQRziaBKT3yNrfv4K9wsrCu9H8dS4OKH

adi160885, 2019-12-16:
Thanks sir

marceltech, 2020-01-06:
> link photo
> https://encrypted-tbn0.gstatic.com/images?q=tbn%3AANd9GcTyPjw2TzDluevxNvkuyQRziaBKT3yNrfv4K9wsrCu9H8dS4OKH
> 
> https://i.stack.imgur.com/wvSuY.png

Hello,

How to use a nodemcu + 3v relay?

kacitran, 2020-01-09:
> > link photo
> > https://encrypted-tbn0.gstatic.com/images?q=tbn%3AANd9GcTyPjw2TzDluevxNvkuyQRziaBKT3yNrfv4K9wsrCu9H8dS4OKH
> > https://i.stack.imgur.com/wvSuY.png
> 
> Hello,
> 
> How to use a nodemcu + 3v relay?

I do not understand your question?
It has a 3v relay, and you want it to be operated with pin nodemcu?
You need an electronic circuit to activate the relay.
Google search you will find many options.
[image]
https://www.geekering.com/?p=187

omersiar, 2020-01-20:
Closing it for now. Feel free to re-open if you need further help with esp-rfid software.

closed 2020-01-20 by omersiar