omersiar/esp-rfid #345: RDM6300 no es capaz de leer

kacitran, 2019-12-15:
the debug firmware 1.0.2 connected to gpio-14 the RDM6300 and shows nothing on the console via RS232 when a card was passed.
Nor does it respond DOES NOT show any type of information access records, such as event records.
the signal arrives at the indicated pin and if I read the information to Gpio-14 it would be the following  31005DA05D91

Any idea what is happening without charge my own firmware without decoding me.

kacitran, 2019-12-15:
ESP8266 -> ESP12E
Gpio-14 It should be low to receive the signal 9600.8.1.N
Once the RDM6300 was disconnected from the Gpio14, it was taken with the Gpio14 oscilloscope showing output signal
[image]
[image]
I can't find the GPIO input and output ports in the programming code
I will continue to investigate
log.......

[ INFO ] ESP RFID v1.0.2            
Flash real id:   0016400E
Flash real size: 4194304
 Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.                                 
[ INFO ] Config file found          
{                                                                          
  "command": "configfile",          
  "network": {
    "bssid": "XX:XX:D1:XX:XX:XX",
    "ssid": "600MB",
    "wmode": 0,
    "hide": 0,
    "pswd":  "kacitranpass",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": "",
    "apip": "192.168.4.1",
    "apsubnet": "255.255.255.0"
  },
    "hardware": {
    "readerType": 2,
    "wgd0pin": 4,
    "wgd1pin": 5,
    "sspin": 0,
    "rfidgain": 32,
    "wifipin": 255,
    "rtype": 1,
    "ltype": 0,
    "rpin": 4,
    "rtime": 400,
    "buttonpin": 255
  },
                                                                          
  "general": {
    "hostnm": "kacitran-Tag",
    "restart": 0,
    "pswd": "admin"
  },
  "mqtt": {
    "enabled": 0,
    "host": "",
    "port": 1883,
    "topic": "",
    "user": "",
    "pswd": ""
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": 30,
    "timezone": 1
  }
}
[ INFO ] Trying to setup RFID Hardware
[ INFO ] WiFi STA Connected
[ INFO ] Trying to connect WiFi: 600MB
[ INFO ] Client IP address: 192.168.1.41
[ INFO ] Trying to setup NTP Server
[ INFO ] Configuration done.

irqnet, 2020-01-16:
Maybe I've got the same issue.

RDM6300 connected to a NodeMCU, GPIO14 (D5) connected to TX pin. Nothing happend when moving a RFID token to the RDM6300.

I've connected the RDM6300 TX to a Serial USB Adapter RX and checked if I could logg the RFID token.

Successful! Serial Console in Putty showed up the token ID.

irqnet, 2020-01-16:
It's working for me when RDM6300 TX is connected to ESP RX

omersiar, 2020-01-20:
Closing it for now. Feel free to open if you need further help with esp-rfid software.

closed 2020-01-20 by omersiar