omersiar/esp-rfid #346: controling servo

m-mahmoudi, 2019-12-15:
hi
Would you like this feature to be included?
runing a servo motor instead of relay
turning 180 degre and goes to 0 after 5 sec...
.
sometimes ppl using servo and i think its a good option to add

daphnaie, 2020-01-09:
Hi,
I think this is a very good idea.
Also button switch to control servo from inside and outside door.

omersiar, 2020-01-20:
Servos in general needs to be in special positions (steps) to unlock or re-lock and there is no programmable -at least for me- for one size fits all logic. At this point you can do it with another small microcontroller (like attiny85) just for listen to esp-rfid relay events and do the servo controlling.

Closing this now.

closed 2020-01-20 by omersiar