omersiar/esp-rfid #347: ESP-RFID Blue Relay Board

nardev, 2019-12-18:
I don't know if you have seen this one. It's much better pinout. Suitable for RC522 and other modules.

> ESP-RFID Relay Blue Board
> 
> Because it was often requested, we made another version. Much more stable, with circuit protections and with better wiring so that it’s possible to use other popular RFID readers like **RC522, PN532, and ofcourse, Wiegand readers**. Below picture regarding the pin usage is pretty self explanatory and if you need any help, feel free to contact us.
> 
> If you haven't seen ESP-RFID project before, you can check the official github page: https://github.com/esprfid
> 
> ESP-RFID Relay Board basically makes any RFID reader wifi enabled and provides several ways to communicate with the reader, manage user cards and log access for the users.

https://www.tindie.com/products/nardev/esp-rfid-relay-blue-board/

[image]

The Blue board's wiring is explained below:

[image]

[image]

The video is coming soon.

teriall, 2019-12-20:
Wow that's nice. Didn't posted it here, but somebody find it already. It's my version on the board. If you know how to handle with a solder iron is possible to make the board yourself. Here is the project: https://easyeda.com/teria/hostel-lock

nardev, 2019-12-21:
> Wow that's nice. Didn't posted it here, but somebody found it already. It's my version on the board. If you know how to handle with a solder iron is possible to make the board yourself. Here is the project: https://easyeda.com/teria/hostel-lock
> And I recommend you to provide the link to the original project at tindie.com coz I developed the project is under MIT licence.

@teriall Man, i was looking for you, but you are so secretive i couldn't find your email. Please contact me. vedran@nardev.org or what ever other way.