omersiar/esp-rfid #348: Lost mqtt communication after 3-4 hours

ojkroken, 2020-01-01:
hello . i get the esp-rfid to work . it work good maybe 3-4 hours ,, and then the mqtt communication from the unit stop sending. it restarts sending when i restart the unit . have sombody a solution  to this or know why i have this issue?
 
the firmvare i use is 1.3.1
the reader: rc-522
card: wemos d1 mini

/best regards pehr

drthanwho, 2020-01-05:
Also experiencing the same issue with the latest version. MQTT suddenly just stops working until a reboot

omersiar, 2020-01-06:
Can you guys collect events regarding the MQTT on event logs ?

ingeninge, 2020-02-03:
Version 1.3.3 can confirm, but the time until disconnect is much longer. Only MQTT Disconnect, no WiFi Disconnection. Velleman RFID Reader and Joy-It ESP8266 Dev Kitv2 4MB Flash. 180seconds
Webserver is running, I can log into esp-rfid, but the card does not work. I can't see a mqtt-message, when I put a card on the reader. Even though access was granted. Works fine after reboot.

BOFH90, 2020-02-21:
i can also confirm this. No mqtt messages after 3-4 hours

agmlego, 2020-02-22:
Confirmed on several different boards. Nothing in event log regarding MQTT.

nardev, 2020-03-15:
The short simple fix should be to add one more connection.

nardev, 2020-03-18:
Maybe your wifi gets reconnected and mqtt doesn't reconnect. 

there is a like in wifi.esp which is commented out and it says:

    mqttReconnectTimer.detach()

Which is supposed to reconnect once the wifi is back on.

nardev, 2020-03-18:
@ojkroken

WAVez86, 2020-03-30:
The same problem here. 
Sometimes it runs for a few days, but most of the time mqtt disconnects after maximum 1 hour
it would be nice to add the fix of nardev into the next version!

ingeninge, 2020-04-05:
> Maybe your wifi gets reconnected and mqtt doesn't reconnect.
> 
> there is a like in wifi.esp which is commented out and it says:
> 
> ```
> mqttReconnectTimer.detach()
> ```
> 
> Which is supposed to reconnect once the wifi is back on.

did not work in dev-tree 1.3.7 for me.
After 8h "Disconected from MQTT server" in event log.

ingeninge, 2020-04-05:
I belive the problem is the timing. You are trying to reconnect when Wifi is connected in wifi.esp: 

void onWifiConnect(const WiFiEventStationModeConnected &event) ...

but that does not mean a valid IP is given. I added a function and a handler in 1.3.3:
wifi.esp:
```
void onWifiGotIP(const WiFiEventStationModeGotIP &event)
{
	wifiFlag = true;
	wifiLedOn();
#ifdef DEBUG
	Serial.println(F("\n[ INFO ] WiFi STA Connected"));
#endif
	//mqttReconnectTimer.detach(); detach destroys the Timer, you wont be able to start a reconnectTimer without initializing an object
	connectToMqtt();
	writeEvent("INFO", "wifi", "Connected with IP", "onWifiGotIP");
}

```
main.cpp:
```

WiFiEventHandler wifiDisconnectHandler, wifiConnectHandler, wifiOnStationModeGotIPHandler;

wifiOnStationModeGotIPHandler = WiFi.onStationModeGotIP(onWifiGotIP);
```

I only tested it once by disableing wifi on my router, but it was the first time the mqtt connection recovered with this little "patch".

christofkac, 2020-04-13:
Integrated as part of issue #374 on dev. 
Downloaded yesterday and have still a running mqtt connection.

MaierA19, 2020-04-18:
In wich Version is the mqtt disconnect fixe include? I have test Version 1.3.1, Version 1.3.3, Version 1.3.5 and in all Version, i have the mqtt disconnect Problems.

ingeninge, 2020-04-19:
Do you want a precompiled version? Until now the fix is only for the source code. 
I downloaded 1.3.3 source code from Release-Tab and got it working with the fix. But it should work with 1.3.1 (dev-tree) since there are no major changes.

MaierA19, 2020-04-19:
Yes please 😅 
Because i will test the System with Hassio

ingeninge, 2020-04-19:
Can you try this please:
[https://github.com/ingeninge/esp-rfid/releases/tag/1.3.3](https://github.com/ingeninge/esp-rfid/releases/tag/1.3.3)

MaierA19, 2020-04-26:
Thanks for the Version, i have test this Version, but i have problems with the mqtt communication. When i reboot the System, the MQTT Server arrived a json Message with the Reboot info, but when i scan a rfid Card, the MQTT Server became no Message.
Sry for my bad English

ingeninge, 2020-05-01:
It does work here. My setup is a bit special, but what you see here ist the MQTT communication between the esp-rfid and the mosquitto MQTT-Broker on mastermind:
`ingmar@ingmar-Aspire-E5-571:~$ mosquitto_sub -h mastermind -t rfid/#
{"type":"heartbeat","time":1588341817,"ip":"192.168.2.72","door":"esp-rfid"}
{"type":"boot","time":1588341829,"Wifi SSID":"JoernGF","Local IP":"192.168.2.72"}
{"type":"heartbeat","time":1588341829,"ip":"192.168.2.72","door":"esp-rfid"}
{"type":"access","time":1588341846,"isKnown":"true","access":"Always","username":"Ingmar_Magic","uid":"4b866aafb5c80","door":"esp-rfid"}
{"type":"heartbeat","time":1588341860,"ip":"192.168.2.72","door":"esp-rfid"}
`
I then let node-red open the door.

MaierA19, 2020-05-01:
Ok thanks
I will Test the Setup sehen i'm home in 3 Day

MaierA19, 2020-05-16:
My Setting of the ESP
{
  "command": "configfile",
  "network": {
    "bssid": "*********",
    "ssid": "********",
    "wmode": 0,
    "hide": 0,
    "pswd": "***********",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": ""
  },
  "hardware": {
    "readerType": 1,
    "wgd0pin": 4,
    "wgd1pin": 5,
    "sspin": 15,
    "rfidgain": 96,
    "rtype": 1,
    "rpin": 4,
    "rtime": 1000,
    "wifipin": 2,
    "doorstatpin": 255,
    "readertype": 0,
    "ltype": 0,
    "openlockpin": null
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": 0,
    "pswd": "admin"
  },
  "mqtt": {
    "enabled": 1,
    "host": "192.168.192.109",
    "port": 1883,
    "topic": "rfid_tuer",
    "user": "mqtt",
    "pswd": "**********",
    "syncrate": "1",
    "mqttlog": 0
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": 30,
    "timezone": 0
  }
}
 The ESP send only this Data to the MQTT Server

{
    "type": "boot",
    "time": 1589617595,
    "Wifi SSID": "ADG 15",
    "Local IP": "192.168.192.191"
}

Acess Data will not send to the MQTT Server an this is my Problem

ingeninge, 2020-05-21:
`{
  "command": "configfile",
  "network": {
    "bssid": "7C:FF:4D:xx:xx:xx",
    "ssid": "MySSID",
    "wmode": 0,
    "hide": 0,
    "pswd": "MySecretPassword",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": "",
    "apip": "192.168.4.1",
    "apsubnet": "255.255.255.0"
  },

  "hardware": {
    "readertype": 0,
    "wgd0pin": 4,
    "wgd1pin": 5,
    "sspin": 15,
    "rfidgain": 32,
    "wifipin": 2,
    "rtype": 1,
    "ltype": 0,
    "rpin": 4,
    "rtime": 800,
    "buttonpin": 255,
    "doorstatpin": 255,
    "openlockpin": null,
    "numrelays": 1,
    "relay2": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },

    "relay3": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },

    "relay4": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    }
  },

  "general": {
    "hostnm": "esp-rfid",
    "restart": 0,
    "pswd": "MySecretLogin"
  },

  "mqtt": {
    "enabled": 1,
    "host": "192.168.2.78",
    "port": 1883,
    "topic": "rfid",
    "user": "",
    "pswd": "",
    "syncrate": "30",
    "mqttlog": 0
  },

  "ntp": {
    "server": "192.168.2.1",
    "interval": 30,
    "timezone": 1
  }
}`

`ingmar@ingmar-Aspire-E5-571:~/$ mosquitto_sub -h mastermind -t rfid/#
{"type":"heartbeat","time":1590059351,"ip":"192.168.2.72","door":"esp-rfid"}
{"type":"access","time":1590059355,"isKnown":"true","access":"Always","username":"Ingmar_Magic","uid":"4b866aaxxxxx","door":"esp-rfid"}
{"type":"heartbeat","time":1590059382,"ip":"192.168.2.72","door":"esp-rfid"}
{"type":"heartbeat","time":1590059413,"ip":"192.168.2.72","door":"esp-rfid"}
`

bad-gargoyle, 2020-08-24:
> Can you try this please:
> https://github.com/ingeninge/esp-rfid/releases/tag/1.3.3

I can only see "blank4mb.bin" in bin folder.

pol-henri, 2020-09-13:
Hi there i have also the mqtt problem, I'm not a king for editing and fix the issue i don't understand realy how , some people can explain or give a link for the how to?  Thanks for your help

C-Gotti, 2021-04-22:
I saw something interessting: 
I had a PiHole running at home. I tried diffrent firmware versions of ESP-RFID and the MQTT always stopped working. I setup a Shelly with IOBroker to restart the ESP-RFID if the MQTT Sync isn't available anymore. 
With PiHole and the local DNS it stopped working after a few hours/days. 

After i disable the PiHole and the router does the DNS again, the MQTT never stopped working again (since 4 weeks). I can't say why, but it looks like that the ESP isn't compatible with PiHole

matjack1, 2022-02-03:
In the `dev` branch the reconnection to MQTT on Wifi reconnection is implemented and also if the MQTT connection gets disconnected, it reconnects automatically after 60 seconds.

Moreover I've recently fixed a bug with reading MQTT messages: https://github.com/esprfid/esp-rfid/pull/475 that was causing to me severe issues.

If you still have problems on the `dev` branch, try running the debug build and report the logs here and I'll try helping :)

matjack1, 2022-03-23:
I'll close this for now as I didn't have any feedback and I think it's now fixed. But if anyone has any issue, please reopen.

closed 2022-03-23 by matjack1