omersiar/esp-rfid #350: Please read before posting issues

frenchie71, 2020-01-08:
Hi chaps,

a lot of people are here to help you - however, sometimes it's a tough call because information is missing in what you describe to be an issue. Please make sure that you include the following information when posting an issue:

- Description of the hardware you are using
- The version of the software that you are using
- **_exact_** steps to reproduce the issue

That would speed up things for us to help you and in a next step include a fix in the code

Many thanks and btw - happy new year ;-)

omersiar, 2020-01-08:
Happy new year 🥂 

https://help.github.com/en/github/building-a-strong-community/configuring-issue-templates-for-your-repository

drthanwho, 2020-02-11:
I was gonna suggest that too. Templates are definitely the best solution, then you make sure people give you as much information as possible to help them out

I still don't see them implemented though 😏👀

matjack1, 2022-10-22:
@omersiar I would like to setup the issue templates in order to make it easier to open issues and to remove this one.

Following the docs: https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository I guess I need some additional permission? Can you help me, please?

closed 2026-10-01 by matjack1