omersiar/esp-rfid #351: Flash.bat not working

TomDrifter, 2020-01-09:
Hi, il be honest im new to the ESP32 and purchased one just for this project. I have spent days trying to get this working and have had to put this on. 

in the arduino software i can upload simple stuff etc like blinking light/wifi scan etc and see this is working on the serial monitor.

i have downloaded the project and gone to the flash.bat file. Opened it, i press 1 and then type COM10 and press enter, the dialog box instantly dispaears. I tried differnt ways (holding down the button on the esp32 for 3 seconds before pressing enter after typing COM10 etc) 

if i press 3 and then type COM10 the dialog stays but shows errors etc and not connecting.

Any help would really be appreciated, Many Thanks

omersiar, 2020-01-10:
ESP32 is not supported at the moment.

TomDrifter, 2020-01-19:
ok so i have now purchased a 8266 i believe but have the same issues. do i need to do anything from the unit being completely new ? any tutorial etc or video on how to install this ?

omersiar, 2020-01-19:
What is the unit you have purchased? Does it have Serial controller? Did you install the serial controller driver?

TomDrifter, 2020-01-19:
Hi, on the chip it says ESP8266MOD, on the bottom it says NodeMCU, Amica, says to install the CP2102 driver, use 9600 baud rate and then connect with wifi. i have installed that driver

omersiar, 2020-01-19:
If the driver is installed correctly, you should note down which port number does it has. Then flash.bat should work. Go to Device Manager and look for the port. 
[image]

omersiar, 2020-01-19:
Some NodeMCU modules are shipping with CH304 Serial Controller, if this is the case you should install CH340 driver instead of CP2102

https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/all

TomDrifter, 2020-01-19:
Thank you so much I will try this as soon as I'm home. Thank you !!

TomDrifter, 2020-01-19:
my hardware is deffo com10 and it says the device is "silicon labs cp210x usb to uart bridge" does this sound right or is that this the driver for my old esp32 ? i appreicate your time

omersiar, 2020-01-19:
What is the error with flash.bat?

TomDrifter, 2020-01-19:
i open flash.bat, get the screen up with options 1,2,3,4. if i select 1 or number 2 and press enter, then type in COM10 and press enter the screen just disapears and does nothing

omersiar, 2020-01-19:
[flash.zip](https://github.com/esprfid/esp-rfid/files/4082982/flash.zip)
Can you please run this one?

TomDrifter, 2020-01-19:
when i do it with this it comes up saying "unknown option 1" and then ask's again for the com port ?

TomDrifter, 2020-01-19:
then when i type the com port again it says "esptool.exe is not recognized as internal external  command, operal program or abtch file"

omersiar, 2020-01-19:
esptool.exe flash.bat and all other files should be in the same folder.

TomDrifter, 2020-01-19:
ok so i copied that file into the original folder and now get this

esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to nodemcu
        setting baudrate from 115200 to 921600
        setting port from COM1 to COM10
        setting address from 0x00000000 to 0x00000000
        espcomm_upload_file
error: stat generic.bin failed: No such file or directory
Press any key to continue . . .

omersiar, 2020-01-19:
generic.bin file should also be in the same folder

TomDrifter, 2020-01-19:
i dont have that file, i have teamvieiwer if you are availble to look and see what i have done wrong ?

TomDrifter, 2020-01-19:
i have blank4mb.bin ?

omersiar, 2020-01-19:
You can download from GitHub, go to "releases" page

TomDrifter, 2020-01-19:
i think you have solved my problem now mate, wow its simple when you know how lol . please let me say my thanks to you on this. i really do appreciate your time to help me. Many thanks !!!

closed 2020-01-19 by TomDrifter