omersiar/esp-rfid #352: More configurable outputs on the code for managing alarm system and anti-robbery

evazzoler, 2020-01-13:
Hi all,
I think my  need can be useful to others for both home and office access control.
I need to connect 2 more relays (through I2C or unused GPIO), this could be an additional optional card or something purchasable on the market.

Relays would serve:
```
A) to deactivate the security system
B) to activate a silent anti-robbery alarm
```

How do they work:

**A)** if the user is enabled to deactivate the alarm (maybe a checkbox property of the user), by typing his PIN (I foresee that the system can only be used with PIN, without RFID tag) or by using his own RFID card, he will deactivate the alarm and open the door.
If the user is not enabled to deactivate the alarm, he can only open the door if the alarm has been deactivated (requires an input for the alarm status or the use of NO / NC for the alarm relay and an alarm system code that only arms)

**B)** if the user under threat enters an emergency code, the door opens and the third relay is activated, causing a silent alarm

TomDrifter, 2020-01-20:
something like have a check box in the user edit menu with "when granted access also trigger gpio ...." etc ?

evazzoler, 2020-01-21:
> something like have a check box in the user edit menu with "when granted access also trigger gpio ...." etc ?

Something similar. The possibility to choose different (single or multiple) GPIO triggered by a user and request a state to be true or false before triggering a GPIO.

I.e.: Tom plays his code and triggers GPIO_A (disarm alarm system) and GPIO_B (door contact).
Jerry plays his code and if alarm system is armed does nothig (receive a green/red/green/red led blink as warning), if alarm system is disarmed the door opens.

Translated in other point of view: Jerry can open door only if alarm is disabled. Tom can always open door.