omersiar/esp-rfid #353: WebGUI does not save Values

theChris1310, 2020-01-16:
I bought the hole board via [Tindie](https://www.tindie.com/products/nardev/esp-rfid-relay-board-12v-for-esp8266-board/) with everything installed.
I can connect via safari and see all the Menus. but 2 things I tried didn't work

- change the WiFi to my home Wifi
- adding new User

Change the WiFi shows the right settings in JSON Window, but after the reboot everything is the same as before.
When I Add a new User, my reader tells and error, but as well I can't see any log entries. Thats why I wanted to get into my network to check the update as I do not have the nesseserry programmer and can't get the Ardoino IDE running on my MacOS *ggggrrrr*

any Idea what I can do so far? Do I see the Version number in the WebGui?

Cheers, Chris

omersiar, 2020-01-20:
Can you please try setting up your device on another browser like Firefox or Chrome?

theChris1310, 2020-01-20:
@omersiar , awesome. That worked. Tested on Chrome for mac Version 79.0.3945.117

Safari version 13.0.4 just loads the left menu. The main page is empty or the forms does not work. 

Closed for me. But if you want, i can test new versions if it helps.

LabVolumerie, 2020-02-24:
same issues for me under safari 13.0.5, under chrome Version 79.0.3945.117 (Official Build) (64 bits) no worries

bsmeding, 2020-04-15:
Same issue here, settings where not saved when using Safari. Moved to Firefox no problem, settings saved and connected to my wifi network now.

jello1974, 2020-04-23:
Is not then doesn't save the settings but some times you must save twice....
Tested on chrome and ie

matjack1, 2022-03-18:
I'm going to close this in favour of this one: https://github.com/esprfid/esp-rfid/issues/302 that has more details, I think the issue is the same.

Please be in touch @theChris1310 @LabVolumerie @bsmeding @jello1974 if you are still interested in this.

closed 2022-03-18 by matjack1