omersiar/esp-rfid #356: Web GUI stops working after a minuite or so

TomDrifter, 2020-01-20:
Heya, i am impressed with this project but have hit a issue i cant seem to fix. 

when i boot the esp up i can log into the web gui easily, add cards, read events etc. but after about a minuite or so it seems to hang. the device still works fine with cards triggering the gpio i have set etc but events stop showing etc, pressing the restart button or unplugging and plugging back in resovles the issue for another minuite. 

i have tried it as a access point and connected to my wifi to eliminate comms issues. 

any help or what to check ? thanks

TomDrifter, 2020-01-21:
ok so i have narrowed this down to mainly the event logs, comes up and hangs on "fetching data"

frenchie71, 2020-01-26:
Hi @TomDrifter , can you please tell us which hardware you are using and which version of the software and how you built it please ? (Downloaded from releases, dev/stable, version number, built with platformio etc.), the reason for my request being that there has been a lot of development in the last 6 months namely on logfiles and maybe you are using an older version? Many thanks.

gabrieluto, 2020-05-28:
hey, i'm having the same issue:
cloned via platformio changed to v1.3.7 and added manually the 4h mqtt loss fix and changed some text to spanish in the frontend, eerytime i go to "log maintenance" and go back to access events, my nodemcu resets.

using nodemcu v0.9(i believe) and a mfrc522

matjack1, 2026-10-01:
This is a duplicate of this one: https://github.com/esprfid/esp-rfid/issues/667 closing

closed 2026-10-01 by matjack1