omersiar/esp-rfid #357: Mqtt commands

wizznet1, 2020-01-23:
Hy all. Im new at esp rfid. I flash my esp with this new version. I managed to connect esp with cloudmqtt...but I dont know how to send messages to add user, delete user, activate relay 1,2,3,4....get door status and if there is some other commands. Please can someon send me commands and tell me how to do it?

_Originally posted by @wizznet1 in https://github.com/esprfid/esp-rfid/issues/343#issuecomment-577403614_

geraldFS, 2020-09-09:
+1 
please!!!

ingeninge, 2020-09-13:
I wrote down some notes about that in my wiki:
https://github.com/ingeninge/esp-rfid/wiki

matjack1, 2026-10-01:
Sorry this went unanswered for so long. All MQTT commands and messages are now documented in [README-MQTT.md](https://github.com/esprfid/esp-rfid/blob/stable/README-MQTT.md). In short: send a JSON message with cmd and the device's doorip to TOPIC/cmd, and replies arrive on TOPIC/send. Adding and deleting users, opening the door, and reading or updating the configuration are all covered there.

Closing. If something's missing from the doc, please open a new issue.

closed 2026-10-01 by matjack1