omersiar/esp-rfid #364: Build Error "error: 'const void*' is not a pointer-to-object type"

Karshilistic, 2020-02-12:
I'm facing this build issue using PlatformIO on VSCode on Windows 10. I tried with 1.0.2 and 1.3.3 and I'm getting the same error.

Before this, I was getting lots of error due to conflict between time.h and Time.h, which I have changed to _Time.h in C:\Users\DIRECTORY\.pio\libdeps\generic\Time_ID44 (as per instructions on some forum). As I can't find anyone else with the same issue, this leads me to believe this is a problem with my installation/setup of PlatformIO or VSCode. Any help or guidance appreciated.

```
Processing generic (platform: espressif8266@2.0.4; framework: arduino; board: esp12e)
----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 2.0.4 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), tool-esptool 1.413.0 (4.13), framework-arduinoespressif8266 2.20500.190223 (2.5.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries
Scanning dependencies...     
Dependency Graph
|-- <ArduinoJson> 5.13.4       
|-- <ESPAsyncTCP> 1.2.2        
|   |-- <ESP8266WiFi> 1.0      
|-- <ESPAsyncUDP>
|-- <ESP Async WebServer> 1.2.3
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 5.13.4
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <MFRC522> 1.4.1 #ea7ee3f
|   |-- <SPI> 1.0
|-- <Wiegand-Protocol-Library-for-Arduino> #812b393
|-- <Time> 1.6
|-- <Bounce2> 2.52
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <Ticker> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio\build\generic\lib6e8\Time_ID44\DateStrings.cpp.o
Compiling .pio\build\generic\FrameworkArduino\core_esp8266_flash_utils.c.o
Compiling .pio\build\generic\FrameworkArduino\core_esp8266_i2s.c.o
Compiling .pio\build\generic\FrameworkArduino\core_esp8266_main.cpp.o
In file included from C:\Users\Karshi\.platformio\packages\framework-arduinoespressif8266@2.20500.190223\tools\sdk\libc\xtensa-lx106-elf\include/string.h:163:0,
                 from C:\Users\Karshi\.platformio\packages\framework-arduinoespressif8266@2.20500.190223\cores\esp8266/Arduino.h:33,
                 from .pio\libdeps\generic\Time_ID44\DateStrings.cpp:12:
Compiling .pio\build\generic\FrameworkArduino\core_esp8266_noniso.c.o
Compiling .pio\build\generic\FrameworkArduino\core_esp8266_phy.c.o
.pio\libdeps\generic\Time_ID44\DateStrings.cpp: In function 'char* monthStr(uint8_t)':
Compiling .pio\build\generic\FrameworkArduino\core_esp8266_postmortem.c.o
CC:\Users\Karshi\.platformio\packages\framework-arduinoespressif8266@2.20500.190223\tools\sdk\libc\xtensa-lx106-elf\include/sys/pgmspace.h:76:81: error: 'const void*' is not a pointer-to-object type
ompiling .pio\build\generic\FrameworkArduino\core_esp8266_si2c.c.o
     #define pgm_read_ptr(addr)              (*reinterpret_cast<const void*>(addr))
                                                                                 ^
C:\Users\Karshi\.platformio\packages\framework-arduinoespressif8266@2.20500.190223\tools\sdk\libc\xtensa-lx106-elf\include/sys/string.h:32:57: note: 
in definition of macro 'strcpy_P'
 #define strcpy_P(dest, src)          strncpy_P((dest), (src), SIZE_IRRELEVANT)
                                                         ^
.pio\libdeps\generic\Time_ID44\DateStrings.cpp:72:29: note: in expansion of macro 'pgm_read_ptr'
     strcpy_P(buffer, (PGM_P)pgm_read_ptr(&(monthNames_P[month])));
                             ^
.pio\libdeps\generic\Time_ID44\DateStrings.cpp: In function 'char* dayStr(uint8_t)':
C:\Users\Karshi\.platformio\packages\framework-arduinoespressif8266@2.20500.190223\tools\sdk\libc\xtensa-lx106-elf\include/sys/pgmspace.h:76:81: error: 'const void*' is not a pointer-to-object type
     #define pgm_read_ptr(addr)              (*reinterpret_cast<const void*>(addr))
                                                                                 ^
C:\Users\Karshi\.platformio\packages\framework-arduinoespressif8266@2.20500.190223\tools\sdk\libc\xtensa-lx106-elf\include/sys/string.h:32:57: note: 
in definition of macro 'strcpy_P'
 #define strcpy_P(dest, src)          strncpy_P((dest), (src), SIZE_IRRELEVANT)
                                                         ^
.pio\libdeps\generic\Time_ID44\DateStrings.cpp:86:28: note: in expansion of macro 'pgm_read_ptr'
    strcpy_P(buffer, (PGM_P)pgm_read_ptr(&(dayNames_P[day])));
                            ^
*** [.pio\build\generic\lib6e8\Time_ID44\DateStrings.cpp.o] Error 1
============================================================ [FAILED] Took 6.12 seconds ============================================================

Environment    Status    Duration
-------------  --------  ------------
generic        FAILED    00:00:06.123
relayboard     IGNORED
debug          IGNORED
====================================================== 1 failed, 0 succeeded in 00:00:06.123 ======================================================
```

Karshilistic, 2020-02-17:
For those curious, I fixed the problem by following this issue: https://github.com/esp8266/Arduino/issues/5733

It now compiles after changing 
`#define pgm_read_ptr(addr)              (*reinterpret_cast<const void*>(addr))` with 
`#define pgm_read_ptr(addr) 				(*reinterpret_cast<const void* const *>(addr))`
in `pgmspace.h` (see original log for full path)

nardev, 2020-03-15:
@Karshilistic i'm bit confused about this one. It doesn't appear in default building the fw but it does if you copy the environment settings, same as global or relayboard and than try to build it, just by renaming the env: value.

Also, i noticed that the default platform without specifying certain 2.0.4 arduino version actually now works but the upload speed needs to be much lower.

so when i changed platform = espressif8266@2.0.4 to platform = espressif8266 in platform.ini it worked.

Karshilistic, 2020-07-16:
Confirming that changing `platform = espressif8266@2.0.4` to `platform = espressif8266` in `platformio.ini` works.

closed 2020-07-16 by Karshilistic