omersiar/esp-rfid #365: ESP File Type

dj-curls, 2020-02-12:
Could someone shed some light on the .esp file type used in the src folder?

Tried googling, and it's turned up nothing useful. It looks like C++ in there. 

Why are the files defined like this, and how can I get VSCode to understand the content?

ver2go, 2020-02-12:
Add Platform IO to VSCode if you want complete tool chain.
Otherwise change the extension to cpp for the linter then change back.
There may be a way to use the cpp linter with esp extension, but not
certain.
Or use the Arudino IDE

omersiar, 2020-02-13:
It is because, in the beginning esp-rfid was a big monolith code. esp files introduced for better readability, but caused more harm than being useful.

nardev, 2020-03-15:
If you set .esp to pe treated as cpp file, than you would not have any problems.

matjack1, 2022-03-23:
As Nardev said, they are just plain cpp files that you can treat as such in your editor. It's just a custom thing of this project.

closed 2022-03-23 by matjack1