omersiar/esp-rfid #366: Editing Code

TomDrifter, 2020-02-16:
Hi, hope this is in the right area, i am new to github and the esp. i am trying to modify random things of this to get it to work with other self-designed devices oh mine (mainly a wiegand reader i made and want to edit parts so it reads cards the way i want it to)

i have arduino software etc and all the packages downloaded. how to i open the current project so i can look through the code and then re-compile ? any help would be great,, thanks

TomDrifter, 2020-02-25:
no help here ?

nardev, 2020-03-15:
Try to read the man page completely first. Start with installing platformio under atom.

matjack1, 2022-03-23:
hey @TomDrifter I'm closing this for now, but maybe hang out in here: https://gitter.im/esp-rfid/Lobby if you are still interested in knowing something more

closed 2022-03-23 by matjack1