omersiar/esp-rfid #368: Need more wiegand bit-length options

vrelk, 2020-02-21:
I noticed in your keypad code (src/rfid.esp line 41), you are only accepting 26 and 34 bits, otherwise it is a keypad entry. It would be nice to also add in 32, 35, and 64 bit options too.

32: another wiegand standard
35: HID Corporate 1000
64: one of the HID readers spits this out for Mifare cards

For output reference: https://pastebin.com/AcND7C5J (this was done by a HID MultiClass RP40 if I remember correctly)

http://cardinfo.barkweb.com.au/

nardev, 2020-03-15:
Hello,

you would have to upgrade the wiegand library that is used. Once it's upgradedd over there, than the compile will reflect and worn in esp-rfid as you expect.

vrelk, 2020-03-17:
Ok, that looks easy enough. For now I have passed out the older fobs which are 34-bit and am keeping the 35-bit ones in a drawer. Still, might be useful to allow it for future releases.

pvtex, 2024-06-12:
i have changed the WIegand Library to WiegandNG.
now you can use Bits length from 24 to 58 bits.
All you need to do is to set the requiered bit length in the Hardware settings.

you can find the change in cc7b24c

matjack1, 2024-07-16:
Thank you @pvtex I want to get V2 out first then we'll take onboard more changes, otherwise this V2 is never going to happen :)

Surely after the V2 is out we can get the code ported here as well, thank you!

Einstein2150, 2025-07-16:
@pvtex I tested the fork and it works great with the new Wiegand enhancements. The only thing is to switch it to 35 bit Wiegand I have to adjust the value directly in the config. The UI has no simple 35 bit option ;-)

Maybe using a free editable value than a selection-list is the solution...

Good work, thank you!