omersiar/esp-rfid #370: Anyway to turn off 401 Basic Authentication?

jc911, 2020-02-29:
# Issue
I cannot use the Web through my iPhone(safari, chrome, firefox and so many apps).
Like this issue:
#302 https://github.com/esprfid/esp-rfid/issues/302
# Origin
I analysed the request by wireshark.
I found that iphone cannot send Authorization header in the request of http://x.x.x.x/ws, although I already pass the http://x.x.x.x/login before.
[image]
[image]
# Solution
So, is there anyway to turn off 401 Basic? @omersiar 
I think I could find some other way to ensure the security.

ver2go, 2020-02-29:
Can you add your with in request as follows?

http://user:password@192.168.1.101

For example

jc911, 2020-03-01:
> Can you add your with in request as follows? http://user:password@192.168.1.101 For example
> […](#)

I'v tried this way. 
But it's not work.
I also tried to put ws://user:pass@x.x.x.x in a websocket js of a html file. And then use my iphone to visit the html file. But it's also not work.

vrelk, 2020-03-17:
Apparently getting rid of the websockets part of the site is on the to-do list, but I can't see that happen anytime soon. More of an FYI than a "just wait a bit" type thing though.

matjack1, 2022-03-18:
hey @jc911, not sure if you are still interested in this, but I'm able to help here. I don't have an Apple mobile device to test this on, but first, can you please confirm it's still happening on the latest `dev` branch?

I'm sure we can find a fix for this!

matjack1, 2022-03-18:
I'm closing this in favour of this one: https://github.com/esprfid/esp-rfid/issues/302 that has more details, but please be in touch if you are still interested.

closed 2022-03-18 by matjack1

jc911, 2022-04-01:
> I'm closing this in favour of this one: #302 that has more details, but please be in touch if you are still interested.

Yeah!
It was sovled on the latest dev branch! I confirmed it a few days ago!
Thank you very much!!! @matjack1