omersiar/esp-rfid #377: Is there a way to retrigger a scan

eldrik, 2020-04-23:
Hi,

i am using version 1.02 with a rc522 reader.

I hide an esp-rfid reader in the charger housing of my vaccum robot one reader on each floor to identify on which floor the robot is docked for loading the specific robot floor map for the next day.

Actually the last seen rfid card triggers an item on my home control software so I am aware of the floor.

But I also like to force a manual read to determine if there is still a rfid card in front of the reader, is this already possible?

Regards
Eldrik

jello1974, 2020-04-23:
That will hapend if you reboot the reader (rfid reader not the esp.....)

eldrik, 2020-04-23:
There is no way to reboot the rc522 on its own. I could power cycle it wiring the 3,3v power line through an relay but there is no room left for further equipment. So I am looking for a sw solution. :/

jello1974, 2020-04-24:
You only need room for the relay.  From what i understand you do not use the open door relay so you can use it to restart the rc522 trough the opendoor mqtt command. Define the card as no access in order to not trigger the relay when the robot is docking .

omersiar, 2020-04-24:
https://github.com/esprfid/esp-rfid/blob/9469e1033a7d08337c88f3396d27b476f2b56ca8/src/rfid.esp#L18

This makes PICC (rfid tag) to NOT respond further readings (from RC522). This is anti-collusion protection comes from specification. I remember trying to find a workaround for it but unfortunately no hope for detecting a PICC that is still in front of the reader. You should google it, this was 5 years ago.

omersiar, 2020-04-24:
@eldrik By any chance you have a Roborock S5 Max?

eldrik, 2020-04-24:
> @eldrik By any chance you have a Roborock S5 Max?

Hi,
yeah its an Roborock S50 with the floor wiping function I don't know if it is a max version.

I also could toggle the robot to leave the station and to return so the card is read again but I'd like to avoid the noise at around Zero o'clock :D

eldrik, 2020-04-24:
> From what i understand you do not use the open door relay so you can use the relay pin to restart the rc522 trough the opendoor mqtt command. Define the card as no access in order to not trigger the relay when the robot is docking and define the relay as momentary active high and wire the gnd reader pin to it.

yeah I will tinker if this is an practical option thx. Do you mind to post the command that is necessary?

eldrik, 2020-04-24:
> mfrc522.PICC_HaltA();

maybe it was this thread? https://forum.arduino.cc/index.php?topic=263865.0

There are 2 code examples how this could be achieved. 
Hmm I think I will try and flash an alternative software might be enough for my use case
https://github.com/Fl0ca/MQTT-RFID_MRC522

omersiar, 2020-04-24:
Yes, 6 years ago, wow. DrDoom seems to see achieve this

eldrik, 2020-04-24:
I now flashed both esp-rfid readers with an alternative sketch where I used the implemented solution from the arduino forum post in combination with an incoming mqtt message for utilizing  a manual read checking if there is a reader in front.

matjack1, 2026-10-01:
Closing this sorry. Meanwhile we have released v2, so if you are still interested, try with the new version and reopen if it's still valid

closed 2026-10-01 by matjack1