omersiar/esp-rfid #379: unable to debug

s-borrelli, 2020-05-04:
Hi
I have a proble with cheap RC522 that don't send any reading
i would like to debug, but, after flashing debug firmware, i didn't see anything on arduino ide
any suggestion?

I'm opening serial monitor and setting to 9600 baud....what else?

omersiar, 2020-05-04:
Just make sure it's working with the MFRC522 Library Examples.

s-borrelli, 2020-05-04:
ok....also with test nothing

s-borrelli, 2020-05-04:
16:47:44.722 -> 
16:47:44.722 -> SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687
16:47:44.824 -> Scan PICC to see UID and type...

matjack1, 2026-10-01:
Closing this sorry. Meanwhile we have released v2, so if you are still interested, try with the new version and reopen if it's still valid

closed 2026-10-01 by matjack1