omersiar/esp-rfid #384: Cannot edit hardware settings

stmark707, 2020-06-17:
Description of the hardware you are using: ESP8266 NodeMCU CP2102 ESP-12E
The version of the software that you are using: v1.0.2
exact steps to reproduce the issue: just logging into the website cannot change from weigand to RC522 device
[image]

stmark707, 2020-06-17:
Picture of my setup:
[image]
[image]

I am getting some new rfid testers friday, it may be a hardware issue i am unsure at the moment

stmark707, 2020-06-17:
i updated the software to 1.3.1, and I am still not able to change the device type. I will update on Friday to see if a different RFID fixes this issue because i'm not sure how the esp8266 is supposed to detect what type of RFID it is or if even does self detect. 
[image]

omersiar, 2020-06-18:
Since you do not have the official hardware, you should flash generic.bin to your development board.

stmark707, 2020-06-23:
that worked! thank you so much.

closed 2020-06-23 by stmark707