omersiar/esp-rfid #385: Video showing ESP-RFID

frenchie71, 2020-06-23:
Hi chaps, not really an issue - I made a youtube video showing how to implement ESP-RFID

[see here](https://www.youtube.com/watch?v=jKuAQTV5Fts)

enjoy.

peteshoard, 2020-10-08:
Thanks for the video, I'm using 1.3.3 and I can't set a pin code other than a single digit. Is this a feature of the 1.3.7 version or do I just have a inferior wiegand keypad?

Thanks

timknowlden, 2023-01-20:
> Thanks for the video, I'm using 1.3.3 and I can't set a pin code other than a single digit. Is this a feature of the 1.3.7 version or do I just have a inferior wiegand keypad?
> 
> Thanks

I have this same issue, I require atleast a 4 digit pin to make this a viable option.