omersiar/esp-rfid #387: how to get all users list in one mqtt message? not each user in separate message

joelgamal, 2020-06-29:
I am using `{
    cmd:'listusr',
    doorip:'(The ESP-RFID IP of the door to open as String)'
}`

matjack1, 2022-10-22:
@windy54 I was thinking the same thing about running out of memory, but surely we can make the JSON bigger, we just need to check how much bigger we can make it to see if we can maybe send at least pages of users, probably better than one at the time, right?

windy54, 2022-10-22:
I am quite happy with one at a time, it is not something I would do that often . When I tested it I had a delay of 0.1 secs between messages and uploaded 150 members. It took 15 secs which I am happy with on an occasional basis.
Not part of this comment but I have had trouble restoring 150 users from a file, have not had time to investigate .

matjack1, 2022-10-22:
Thank you very much for the feedback!