omersiar/esp-rfid #391: Cant add a new user (tags/card not scanning)

beped, 2020-07-23:
I can reach the user page, the red lend on the RC522 its on. But when i aproach the tag nothing happen.
I have double checked the wires and have tryed another board (W1 wemos mini and NodeMCU), same problem.
Im using the 1.3.1 version.
Any ideais what can it be?
[image]

rasmusoeh, 2020-07-25:
Have the same issue, tried installing older versions but the issue still occured

beped, 2020-07-28:
No one to help us? Its the firmware maybe? i have downloaded from here like 5 different firmwares, same problem.

beped, 2020-07-28:
My setup:
[image]
[image]

beped, 2020-08-13:
none?

beped, 2020-08-26:
Still...

20:12:01.062 -> [ INFO ] Trying to setup RFID Hardware
20:12:01.165 -> [ INFO ] RFID SS_PIN: 5 and Gain Factor: 32
20:12:01.199 -> [ INFO ] MFRC522 Version: 0x0 (unknown)
20:12:01.234 -> [ WARN ] Communication failure, check if MFRC522 properly connected
20:12:01.302 -> [ INFO ] Trying to connect WiFi: HahaHaha
20:12:01.371 -> 
20:12:01.371 -> [ INFO ] WiFi STA Connected
20:12:03.227 -> [ INFO ] Client IP address: 192.168.68.126
20:12:03.262 -> [ INFO ] Trying to setup NTP Server
20:12:03.296 -> [ INFO ] Configuration done.

sepalab, 2020-09-08:
Try to increase the gain to the max 48dB. It should work.

beped, 2020-09-20:
@sepalab already tryed that, didn`t work.

jvcaloen, 2020-11-30:
I have the same issue (red LED on but no scan) and no luck with the MFRC522 library DumpInfo example either. Either I didn't solder the pins right (3 times) or the board has a defect or both included tags don't work. Any advice to further troubleshoot welcome

beped, 2021-01-07:
> I have the same issue (red LED on but no scan) and no luck with the MFRC522 library DumpInfo example either. Either I didn't solder the pins right (3 times) or the board has a defect or both included tags don't work. Any advice to further troubleshoot welcome

Have you find out how to make it work?

jvcaloen, 2021-01-07:
It seems the MFRC522 module was faulty, I got another one which works fine.
One update: the blue keychain tags which come with the mfrc522 don't work.

beped, 2021-01-12:
> It seems the MFRC522 module was faulty, I got another one which works fine.
> One update: the blue keychain tags which come with the mfrc522 don't work.

I have bought like 6 or 7 tags from different sellers here in Brazil. Always the same problem. But its allways the blue keychain and white card who came with the boards. Maybe is that?

jvcaloen, 2021-01-12:
For me the second mfrc522 reader worked with both white cards from the 2 sets I have.

matjack1, 2026-10-01:
Closing this sorry. Meanwhile we have released v2, so if you are still interested, try with the new version and reopen if it's still valid

closed 2026-10-01 by matjack1