omersiar/esp-rfid #392: What is the mqtt topic?

XVD-XVD, 2020-07-25:
Those are the new features.

Added
[firmware] Support for Door Status tracking with log and mqtt options @nardev
[firmware] adding mqtt_publish_info, generateUid functions @nardev
[firmware] MQTT functionality is greatly improved, thanks to @marelab

But I don't know what the topic is can someone help me?

ingeninge, 2020-09-11:
in 1.3.3: it is a JSON string. doorip ist the ip of the ESP-RFID:
mosquitto_pub -h mastermind -t testrfid -m '{"doorip":"192.168.2.98","cmd":"opendoor"}'

nardev, 2020-09-11:
@ingeninge tested? should we merge it?

ingeninge, 2020-09-11:
Yes tested it just a few minutes before with a fresh installation on the Node-MCU.

Mit freundlichem Gruß, Kind regards,
Ingmar Joern

closed 2022-04-06 by XVD-XVD