omersiar/esp-rfid #393: Wiegand stops after a 4 weeks

rodgerlubbers, 2020-08-06:
I made this project on a wemos d1 with build 1.0.2 and a wiegand reader.
It works very well.
Only after about 4 weeks The wiegand reader is no longer read.
The webui still works and with a reboot id the webui everything works again.
Until about 4 weeks have passed.
It looks like the problem "after a while the RFID reader stops # 223"

crs26, 2021-08-20:
I'm using MFRC522 and the reader always stops after some hours, after reboot it works again but after a while, the reader stops working again. I guess an option to reset the reader after some minutes would be handy.

matjack1, 2025-07-27:
hey @mdagbro do you know that you can turn-on the auto reboot on esp-rfid? Or the issue is with the Wiegand reader itself? Maybe the auto-reboot can mitigate your issue?