omersiar/esp-rfid #394: How to flash chances Made on HTML to nodemcu

joni9990, 2020-08-13:
Hi, how can i flash the frontend changes? I've tried node gulpfile.js but i have no results.
Thanks

gabrieluto, 2020-08-23:
"You can not simply edit Web UI files because you will need to convert them to C arrays, which can be done automatically by a gulp script that can be found in tools directory or you can use compiled executables at the same directory as well (for Windows PCs only)."
it's on the ReadMe

matjack1, 2022-03-30:
yes @joni9990 it's now slightly improved since when you tried. If you go in tools/webfilesbuilder, then you install the npm packages with `npm i` and then run `npm start` it should "compile" the web ui and prepare it to be flashed.

I'm going to close this for now, but please reopen if necessary.

closed 2022-03-30 by matjack1

windy54, 2022-07-29:
It would be great if the ReadMe was changed to actually add the instructions matjack1 provided above.
I saw the instructions on the readme but using js before did not what to do.

[if you want to edit esp-rfid's Web UI you will need (unless using compiled executables):

NodeJS
npm (comes with NodeJS installer)
Gulp (can be installed with npm)
Cd to the tools/webfilesbuilder folder
npm i
npm start
and the following filed will be updated ....]

os something similar