omersiar/esp-rfid #395: UID not the same as Tag Number

stoehrmark, 2020-08-13:
Great code. Thanks for all the work on this!!

My question is why are the UID numbers not the same as the number printed on the Tags?

When I scan the tag with the number 1712142 esp-rfid gives me back 15065073  ??

[image]

Pako2, 2020-08-20:
1. There are different ways to evaluate the read value. We cannot know which method the manufacturer has chosen.

2. In fact, it is the same number, only it is bit negated. We can see this when we convert both numbers into a binary number system:
111001011101111111110001 = NOT 000110100010000000001110

I have a similar "problem". The numbers seem different. But when I take the number obtained with esp-rfid, I convert it to hexadecimal, I take the last 3 bytes, I convert it back to decimal, then I get the number that is printed on the RFID tag ...

vrelk, 2020-12-29:
This is because of the large number of different card formats. esp-rfid does it in a generic way. Here is a list of many different card formats so you can see. https://web.archive.org/web/20200225020141/http://cardinfo.barkweb.com.au/ Because of this, unless you know the card format, and are able to tell esp-rfid how to interpret it, the numbers won't match.