omersiar/esp-rfid #396: Captive Portal

Karshilistic, 2020-08-20:
I was trying to wrap my head around implementing captive portal and thought to check if it was ever discussed here, and found the commit history. I noticed that back in 0.3Alpha2 (August 2017) captive portal was quickly removed after it was added. Is there any reason for this? Is this a feature that can be revisited?

omersiar, 2020-08-20:
IMHO captive portal is just a dirty hack, it's not guaranteed to work in next year nor every device supports it. There is no standard around it and it seems there isn't any attempt to standardize it AFAIK and it consumes great amount of precious resources in the microchip.

Karshilistic, 2020-08-20:
Makes sense. It does seem like a lot of resources for a slight convenience. Thanks for the response.

closed 2020-08-20 by Karshilistic