omersiar/esp-rfid #399: Support of Fingerprint readers?

henfri, 2020-08-30:
Hello,

There are Wiegand Readers, that also have Fingerprint Modules, like [this](https://de.aliexpress.com/item/33056522572.html?spm=a2g0o.productlist.0.0.5b4c606eV2rGHo&algo_pvid=b098170a-89be-45c7-8e58-0d503f9b0429&algo_expid=b098170a-89be-45c7-8e58-0d503f9b0429-6&btsid=77871500-9432-44d2-aebf-c2f566d1ce39&ws_ab_test=searchweb0_0,searchweb201602_10,searchweb201603_53) one.

Would this work as well?
I could imagine, that this Device behaves as if a Keycard is held in front -but I am of course not sure.

Does anyone have Experience with this?

Regards,
Hendrik

ozett, 2020-09-28:
i have such fingerprint-readers running through wiegand Do/D1 on esp's with firmware "espeasy".
running here like RFID or any other reader, as long as they are connected via Wiegand D0/D1 wires (is 5V).

fabiosoft, 2020-09-30:
Same for me... @ozett do you "hardcode" the passkeys inside espeasy somehow or you depend on same sort of host for code validation?

This project seems very promising for my use case... (and also interested on espeasy... i never used)

ozett, 2020-09-30:
@fabiosoft 
the key-id is sent from esp-easy via MQTT to node-red.
there i check and allow/deny the door-openings within a timeframe.

[image]

fabiosoft, 2020-10-01:
@ozett very cool implementation! Thanks for your sharing 🥇

roysbike, 2020-10-01:
> @fabiosoft
> the key-id is sent from esp-easy via MQTT to node-red.
> there i check and allow/deny the door-openings within a timeframe.
> 
> [image]

what is fingerprint ? Generic R307 Optical Fingerprint Reader?

henfri, 2020-10-05:
Hello,

thanks for sharing. Did you use anything to change the Voltage? Wiegand needs 5V, whereas ESP8266 uses 3.3V.

Greetings,
Hendrik

ozett, 2020-10-05:
in 2016 i discovered some information that the esp-pins can take 5V (even if they are officially 3.3v).
--> https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=1617

since than may readers runs with 5v on the wiegand pins without levelshifter directly on the esp.
 working fine without any hassle since than

henfri, 2020-10-05:
Hello,

thanks, I will have a look into it.
Currently, I have [this](https://www.aliexpress.com/item/4000436677714.html?spm=a2g0o.productlist.0.0.5aee6a7eosjAea&algo_pvid=8098d5b6-d767-45c2-a005-cae7d464c842&algo_expid=8098d5b6-d767-45c2-a005-cae7d464c842-0&btsid=0b0a0ae216019244787233491ea939&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) reader (I can currently not recommend it, as I get no support at all from the Manufacturer).

I have it connected to a level shifter using the TXS0108E.
[image]

I have connected:
- GND: To the GND of the ESP which is also the GND of the Fingerprint reader
- VA to 5V on ESP module
- VB to 3.3V on ESP module
- A1, A2 to ESP D7, D6
- B1, B2 to Wiegand D0, D1

That is correct, isn't it?
Nevertheless, I do not get anything from the reader in esprfid.

Do you see my mistake?

Regards,
Hendrik

ozett, 2020-10-07:
difficult diagnosis. i had same  difficulties with level-shifter and finding an error, so i decided to risk one esp-board and connected the reader directly.

maybe you try this? 
get espeasy, the download includes the flashing tool.

wire power the reader with 12v,
wire D0/D1 (5V) to two esp-pins directly. 
configure on espeasy a device for  wiegand-reader 
and check log on the espeasy modul if there is key-id coming up.

or use an oszilloscope for D1/D0.. ?

by the way:
How fast is the fingerprint reading on this reader? i considered to buy one...

henfri, 2020-10-13:
Hello,

it works in principle (direct connection to wemos). Thanks for your hint.
The problem is, that as soon as I put the reader (S4A TF1) into Reader (opposed to standalone mode), I cannot access the Admin mode anymore. 
Not much lock with the support yet.
The fingerprint reading is fast though.

Regards,
Hendrik

ozett, 2020-10-13:
i dont looked up the manual. the reader seems to have wifi, too?

my reader "always"! sends card-id through  wiegand. (even the readers dont allow access)
with fingerprint, it sends only finger-id if the finger is allowed.

and i put the reader only for some  minutes in admin-mode to read new finger.
after that the reader stays in reader-mode (nomal-mode) for years..

henfri, 2020-10-13:
Hello,

the Wifi is ridiculous. You can only open/close the lock. I was expecting to be able to manage Cards / Fingerprints as well, as possible for other readers (e.g. [this ](https://de.elv.com/elv-fingerprint-zahlenschloss-fp100-251211)one).

I have not yet checked cards. For fingerprints it is the same behaviour as you describe.

>and i put the reader only for some minutes in admin-mode to read new finger.

Same here. 

>after that the reader stays in reader-mode (nomal-mode) for years..

Readermode here is not the counterpart to Admin-Mode.

The Device has three Modes: 
1) Standalone Access Control Mode
2) Relay Toggle Mode
3) Reader Mode

In 1&2 everything is done by the device itself, not via Wiegand.

So I need Mode 3. But as soon as I switch to that, I do not get into the admin mode anymore.

Regards,
Hendrik

ozett, 2020-10-13:
is there no output on wiegand in mode 1 or 2 ? did you check?

anyways: can you acitvate admin-mode with a card or finger? 
or only via wifi? 
or never again after going into reader-mode?...

luby1, 2021-05-07:
Hendrik, did you manage to resolve this? I am looking at buying the same keypad. Looking at the manual, in reader mode it seems you can only add fingerprints. So I imagine that if you add them by user id, then when you touch a fingerprint sensor, the user id would be sent via wiegand interface. Whereas when you touch the keypad by the RFID card, it should just send RFID card id via wiegand interface, and lastly if you type in a code on the keypad, in the reader mode it should just pass what you type via wiegand interface. This is probably why you can't add users in reader mode, as keypad is just passing what it reads.
Are you able to confirm that TF1 is behaving in reader mode as I described? Thank you.

fabiosoft, 2021-05-28:
I have a different reader (but probably inside are the same) with keypad and fingerprint 
It works all independently and broadcast Wiegand information only for already registered fingerprints. 
it works standalone so I don’t  know how useful is add to Wi-Fi  network if I have to manually add and delete allowed people. 

Any suggestions? I am open to solder wires if needed ahah 😂

ozett, 2021-05-29:
what model? links?

suggestion: use reader `wired` to the esp-easy -- and than use esp-easy `wireless `to trigger door-opening with locks. 🧷

ypoosn, 2021-11-20:
> Hello,
> 
> the Wifi is ridiculous. You can only open/close the lock. I was expecting to be able to manage Cards / Fingerprints as well, as possible for other readers (e.g. [this ](https://de.elv.com/elv-fingerprint-zahlenschloss-fp100-251211)one).
> 
> I have not yet checked cards. For fingerprints it is the same behaviour as you describe.
> 
> > and i put the reader only for some minutes in admin-mode to read new finger.
> 
> Same here.
> 
> > after that the reader stays in reader-mode (nomal-mode) for years..
> 
> Readermode here is not the counterpart to Admin-Mode.
> 
> The Device has three Modes:
> 
> 1. Standalone Access Control Mode
> 2. Relay Toggle Mode
> 3. Reader Mode
> 
> In 1&2 everything is done by the device itself, not via Wiegand.
> 
> So I need Mode 3. But as soon as I switch to that, I do not get into the admin mode (the one to add users) anymore.
> 
> Regards, Hendrik

@henfri : It is not documented properly in the manual you have to really long press the * key until you get a beep followed by the mastercode and the # key when device is in Wiegand reader mode. 

best Ypo

henfri, 2021-11-27:
Wow thanks. That is something that the tech-support could not tell me. I will give it a  try!

How did you find out?

luby1, 2021-12-18:
Did anyone find issues with S4A (tf1) in reader mode with virtual card number output setting (menu 532#)? This mode is the only one that passes actual code entered on the keypad and also the id of the programmed fingerprint. I find that despite the light on the keypad going green after entering 4 digit pin, it doesn't send anything to the controller (in my case Fanvil i32V) at times. Sometimes only 3rd attempt sends the code (or fingerprint).  I am pretty sure it is not Fanvil unit, because if it gets something it can recognise, irrespective of whether it is the code that is programmed for access or not, it ends up in the log, but in cases when S4A reader lights the green light and nothing happens, there is no entry in the Fanvil log. Only the mifare tags have 100% reliable transfer of data to Fanvil unit, so it must be either a defect in my unit, or some interference that makes data completely invalid for Fanvil unit so it doesn't see it even as failed access.

henfri, 2022-02-07:
Hello,

> @henfri : It is not documented properly in the TF1 S4A manual. You have to really long press the * key until you get a beep followed by the mastercode and the # key. Then device is in admin mode when Wiegand reader mode is active.

I confirm this. Thanks a lot! Unbelievable that the Support could not tell me this.
Again, how did you find out @ypoosn  ?

Best regards,
Hendrik

henfri, 2022-06-19:
Hello,

I found something very concerning:
I have learned several Fingers. When a finger is now pressed on the reader, it sends the UID - starting with 1 - to the reader.
[image]
So far so good.

But now, if I enter just 1# via the keypad, the lock also opens - as if I had presented my finger.
`Jun 19 18:54:24 esp-rfid/send {"cmd":"log","type":"access","time":1655657664,"isKnown":"true","access":"Always","username":"Hendrik li zei","uid":"1","door":"esp-rfid"}
`

Wow.. So, the only thing one needs to do in order to overcome the fingerprint is to enter random IDs starting with 1 followed by # ?!

Any ideas how to fix this?

henfri, 2022-06-23:
> Did anyone find issues with S4A (tf1) in reader mode with virtual card number output setting (menu 532#)? This mode is the only one that passes actual code entered on the keypad and also the id of [...]

Sorry, I missed this one.
I have the same problem. 
For all three options (Key, Finger, Card), the device seems to send only approximately evey third time something over wiegand.
@luby1 were you able to solve this?

Best regards,
Hendrik

luby1, 2022-06-25:
> Sorry, I missed this one. I have the same problem. For all three options (Key, Finger, Card), the device seems to send only approximately evey third time something over wiegand. @luby1 were you able [...]

Hi Hendrik, while I confirmed it is the unit itself rather than Fanvil video phone/controller, after spending too much time on the various configurations, I gave up at the end.  

During initial testing when I had short cables and everything was inside, I never had a problem. Maybe it was pure luck then, but I do think it has something to do with stable temperature and/or air moisture level. For example, for the past 2 days, every time I used it, it worked the first time. I was using it when outdoor temperature was around 15-20 degrees centigrade mark. I also noticed when it fails, if I rubbed the keypad everywhere, then pressed '#' just to clear the buffer, and tried afterwards, it usually works after that.  Whether this warms up the keyboard, or just 'wakes up' the reader from some kind of sleep, I don't know. 

One thing that I didn't look into is the LED signal on Wiegand interface. During my initial testing indoors, I didn't bother connecting it and unit's LED appeared to be working independently fine. But when I had gate people come to install automation with 2 gates, I told them to connect LED wire since this is how fingerprint reader manual suggested it. I'll wait for summertime here in NZ in 6 months when it is dry and then try to open the pillar with wires and disconnect the LED signal, for my final test.  

I also searched for similar unit that has keys you need to press, rather than this touch unit, but most of them I could find have no fingerprint reader and virtual keyboard mode, which I need.

Let me know if you get at the bottom of this issue.

henfri, 2022-06-26:
Hello,

so, you gave up to improve it, but you are still using it, right?
What is your success-rate now? We are above 22°C here currently and the Device is exposed to sunlinght and thus even warmer.
Have you figured out, what the `532`(virtual key) vs `531` and `530` modes mean for the Keypad usage? 
I have the LED signal not connected (only 12V and the two Wiegand wires). 
I was also thinking that the cable-length could be the reason. But Wiegand is supposed to be very stable.

Best regards,
Hendrik

luby1, 2022-06-26:
> Hello,
>
> so, you gave up to improve it, but you are still using it, right? What is your success-rate now? We are above 22°C here currently and the Device is exposed to sunlinght and thus even warmer. [...]

Yes, I use it and have learnt not to be too annoyed when it is not behaving. Just as I wrote that reply to you yesterday, the day became sunny and the keypad was exposed to it, and lo behold it failed first time after 2 days of stability. So I just rubbed the keypad all over, pressed '#' and then I was able to get it to go. Length of wires was my early suspicion but I have no way of proving it now when it's in use.  The standard recommends using shielded wires, but my gate installer used just CAT6. The distance is around 10m, so even without being shielded, it should work, and it does work once you get it going. Once it starts working, I can just keep entering the codes and reader keeps sending them to fanvil with no issues. If length was a problem, then I would have at least intermittent issues even then, but I don't.

You see, when Fanvil gets a wrong code from the keypad (and this happens sometimes), it beeps (due to my configuration) and I can see it in the log. When keypad doesn't work, it actually doesn't send anything at all, since nothing is registered at fanvil. 

Considering once you get it going, it works fine, I tend to think it is more temperature stability related, or perhaps there is a bug in its software where it just goes to sleep. For example, yesterday, the direct sun heated it up a bit, and I think this did something to the membrane or internals of the unit so it was not really doing anything until I rubbed the keypad everywhere first.  So to answer your question, on average I have to intervene every few times I use the keypad. Or can use the tag instead, which seems to work reliably, provided you place the tag where it expects it (close to the fingerprint sensor). For example, if I open the gate when going for a walk with the dog with no issues, generally, I'll open it with no issues upon my return. Also, if I have to fiddle with it to get it to open when I go out, upon my return (30-45 minutes later) generally I also have no issues.

The 531 and 530 modes are for the devices that expect 8 bit burst or 4 bit burst. From what I understand, this is used when you have just a keypad and want to connect it to a mifare card reader which is also an access controller, or gets connected to another access controller. I was unsuccessful in using these modes with fanvil, because it expects wiegand data (32 bit).

I hope this helps.

luby1, 2022-06-27:
> Hello,
>
> so, you gave up to improve it, but you are still using it, right? What is your success-rate now? We are above 22°C here currently and the Device is exposed to sunlinght and thus even warmer. [...]

BTW, I just did more tests with tag, and I can tell you that tag reading is almost 100% reliable. It has to be placed close to "0", and there is an ideal distance from the reader (not too far, but not touching it either). I also found that fingerprint is also sent to fanvil all the time, but obviously sometimes reader doesn't find the match if you don't place the finger properly, and this is indicated by a beep and red light.

So it seems only keypad is having issues in my case. Let me know if you solve this somehow at the end.