omersiar/esp-rfid #400: Relais not switching when pushing "test"

henfri, 2020-08-30:
Hello,

I have tested my relais board with this code:
```
#define RELAY_PIN 5
void setup() {
 pinMode(RELAY_PIN, OUTPUT);
}
void loop() {
 digitalWrite(RELAY_PIN, HIGH);
 delay(1000);
 digitalWrite(RELAY_PIN, LOW);
 delay(1000);
}
```
unfortunately, with the "Test" Button in the Hardware Section, I cannot get the relais to switch.
What could be the reason? 
I have not yet connected a Wiegand Reader - could that be the cause?

I am mildly confused about the different ways to name the pins...
This Pin 5 according to [this ](https://i2.wp.com/randomnerdtutorials.com/wp-content/uploads/2019/05/ESP8266-WeMos-D1-Mini-pinout-gpio-pin.png?w=715&ssl=1)schematic
Could be GPIO5 (=D1) or D5 (=GPIO14).
I have tested both settings in the webinterface (GPIO5 and GPIO14) without success :-(

Regards,
Hendrik

ingeninge, 2020-09-13:
I found the test failing, when you change the Pin in the settings and press Test without saving and restarting. Could depend on ESP-RFID version, but please give it a try. Also my Relay is 5V Arduino-type and I was not able to trigger it stable with 3,3V VCC from the NodeMCU Board on some Pins (GPIO5 fx.). Try D7 (GPIO13) this works for me.

henfri, 2020-09-13:
Hello,

you are right. A reboot was neccessary. 
I think, it would be advisable to trigger that when the settings are saved.

Regards,
Hendrik

Einstein2150, 2025-07-16:
I had problems with my 5v Relay too. The solution was to switch it with a MOSFET to ground - not directly with the esp.

closed 2026-10-01 by matjack1