omersiar/esp-rfid #401: multiple relay's

Tinker-Bel, 2020-09-09:
Hello,
I saw the youtube video demonstrating  ESP-RFID. It mentioned the use of multiple relays. I've installed version 1.3.1 but can't seem to find the parameter to change the number of relays to for instance 2.
Am I missing something? or a wrong version?

Kind regards

ingeninge, 2020-09-14:
Not sure if this is the way, can't check it right now. In dev-Version of ESP-RFID you can change int numRelays=1;
https://github.com/esprfid/esp-rfid/blob/dev/src/main.cpp
Line 44
Give it a try and report here please

ingeninge, 2020-10-07:
Now I checked. You don't need to change anything in the source code. If you compile 1.3.7 you can set Number of Relays to 2. You can set a GPIO to the second Relay
[image]

Then you can add a user, scan the tag and choose witch relay should be activated (default is all). Disable a relay ("Access Type" or "Access Relay 2") if you want the user only to be able to enter one room.
[image]

Tinker-Bel, 2020-10-11:
yes, now I get the confusion!
Multiple relays are in version1.3.7.
I' ve installed the 1.3.7, but the version shown on the webpage is 1.3.1.
That's what been displayed in the youtube video.

thanks for getting me on the right version!!!

closed 2020-10-11 by Tinker-Bel