omersiar/esp-rfid #402: MQTT message stop after some Days

pol-henri, 2020-09-12:
hi, tanhks for helpin advance  at each door event the esp -rfid does send the user, the time exc .., but after some day the MQTT messages stop, the door still opens and the esp is online with IP. I have to restart it for everything to work again, choosing to restart every 24 hours does not correct the problem. I am with version 1.0.4 and it is green, not blue, what i need to do ? there is a upgrade ? a trick ? tanks for your help

ingeninge, 2020-09-13:
Please check if it is related to:
https://github.com/esprfid/esp-rfid/issues/348

pol-henri, 2020-09-13:
Thanks for your help 👍

pol-henri, 2020-09-18:
maebe you know where i can find link for "how to" update firmware ? on the http:// of the module there aren't any update available, but i'm on V1.0.4 and i see on the Link "https://github.com/esprfid/esp-rfid/issues/348 " V1.3.3  , this is not the same ? my module is green , thanks for your time

ingeninge, 2020-09-19:
V1.0.4 seems to be "stable" version. Download generic.bin: https://github.com/ingeninge/esp-rfid/releases/tag/1.3.3
and upload via the GUI menu "Upload" if available in V1.0.4. Else you will have to flash it with the downloaded version.
Note this is not the stable version, but you might find it more stable than other versions available. If so please write a comment about your experiences. Thanks in advance.

pol-henri, 2020-09-19:
Thanks, I try this this weekend, all is stable for me only mqtt message have problem, 7 hope this update fix it, I post my feetback 👍thanks a lot Envoyé depuis mon appareil mobile Samsung.
-------- Message d'origine --------De : Ingmar Joern <notifications@github.com> Date : 19/09/20  12:24  (GMT+01:00) À : esprfid/esp-rfid <esp-rfid@noreply.github.com> Cc : pol-henri <graphxphd@gmail.com>, Author <author@noreply.github.com> Objet : Re: [esprfid/esp-rfid] MQTT message stop after some Days (#402) 
V1.0.4 seems to be "stable" version. Download generic.bin: https://github.com/ingeninge/esp-rfid/releases/tag/1.3.3
and upload via the GUI menu "Upload" if available in V1.0.4. Else you will have to flash it with the downloaded version.
Note this is not the stable version, but you might find it more stable than other versions available. If so please write a comment about your experiences. Thanks in advance.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/esprfid/esp-rfid/issues/402#issuecomment-695195787",
"url": "https://github.com/esprfid/esp-rfid/issues/402#issuecomment-695195787",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

pol-henri, 2020-09-19:
I try with the zip version but when the system as rebooted it still on 1.0.4... Need tu try the bin file or tar? Thanks Envoyé depuis mon appareil mobile Samsung.
-------- Message d'origine --------De : Ingmar Joern <notifications@github.com> Date : 19/09/20  12:24  (GMT+01:00) À : esprfid/esp-rfid <esp-rfid@noreply.github.com> Cc : pol-henri <graphxphd@gmail.com>, Author <author@noreply.github.com> Objet : Re: [esprfid/esp-rfid] MQTT message stop after some Days (#402) 
V1.0.4 seems to be "stable" version. Download generic.bin: https://github.com/ingeninge/esp-rfid/releases/tag/1.3.3
and upload via the GUI menu "Upload" if available in V1.0.4. Else you will have to flash it with the downloaded version.
Note this is not the stable version, but you might find it more stable than other versions available. If so please write a comment about your experiences. Thanks in advance.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/esprfid/esp-rfid/issues/402#issuecomment-695195787",
"url": "https://github.com/esprfid/esp-rfid/issues/402#issuecomment-695195787",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

pol-henri, 2020-09-19:
7 try with zip and tar version, the result still the same, esp still on 1.0.4 firmware

ingeninge, 2020-10-08:
To upgrade from V1.0.4 to higher version it looks like you'll have to erase the flash memory of the esp32. You can do so with esptool.py under linux with erase_flash as argument.

pol-henri, 2020-10-08:
Yes I try to do whith esp-01and it work, so with the esp-rfid  there aren't no mini USB or so, this is with  "Ota" wireless or a add-on card like for esp-01?  Thanks for your time