omersiar/esp-rfid #405: RFID-RC522 stopped working after running 24hours

vdj69, 2020-09-21:
Hi there!
Last week I have received my Official Blue Board and RFID-RC522 stopped working randomly after running 12-24hours but after doing reset/reboot it started working. the issue is repeating again and again.

Any on using this hardware with other popular/stable libraries like Tasmota or Espurna?

ingeninge, 2020-09-28:
This sound much like the issue I had. Please check if it is related to:
#348

amir-hg, 2022-01-03:
Hi,
Did you find any solution?

lazarus72, 2022-01-18:
i also need help for this issue.

> This sound much like the issue I had. Please check if it is related to: #348
i cant confirm, because i don't have PiHole

i use: Tasmota Version | 10.1.0(TasmoCompiler-esp8266generic)

are there any settings to make it stable?