omersiar/esp-rfid #406: Updating last stable 1.0. 4 to 1.3.3 or higher

pol-henri, 2020-09-21:
Someone can explain to me how to update esp-rfid to new firmware, actually I run on 1.0.4 and want to update to 1.3.3, so I use the upload button but esp-rfid still on 1.0.4 after reboot, there is another way to update? Thanks in advance

ingeninge, 2020-09-28:
At least you should be able to flash it with esptools via serial link. I Don't know if you can backup your data from 1.0.4 and restore it in 1.3.3, but you better make the backups before trying. If it fails you can fall back to 1.0.4 
Also consider to make a flash erase if simple overwriting with esptool fails.

nardev, 2020-10-02:
@ingeninge can i get your email, or at least send me one at vedran@nardev.org i would like to pass you some message.

peteshoard, 2020-10-07:
Problem seems to occur when you don't erase the flash ahead of the fresh push. 

Tasmotizer (https://github.com/tasmota/tasmotizer) makes it nice and easy, but 'Flash.bat' in the utils works also

ingeninge, 2020-10-08:
In Linux you can do:
`esptool.py --port /dev/ttyUSB0 erase_flash` and flash with the new version.

pol-henri, 2020-12-02:
thanks for your help

pol-henri, 2020-12-10:
With Tasmotizer abd the 1.3.3.bin file i try to update firmware , i use a TDI232 as UART serial connection , but a c'ant acces to esp , is  a special thing to do ?
here picture how a connect : 
https://ibb.co/w6RpNP7
https://ibb.co/jgR1kpF

thanks if you can help me on this

nardev, 2020-12-10:
@pol-henri standard programming procedure as for any other ESP8266:

1. Press and hold FLASH button
2. Power Up the board or if it was already powered press RESET
3. Release the FLASH button

The blue led on the ESP-12F module should not be blue at that point. The module should be ready for uploading the sketch.

pol-henri, 2020-12-10:
thanks a lot i will try it  👍

pol-henri, 2020-12-11:
back to trying, this time i use the right precedure , but again..can't access to the Esp , i use only the VCC , GND RX, and TX from "UART" connector , i try with easyesp too , same result, so on my new "BLUE" version i try the update via UI , and this time it work !?, so lucky i try to the older RFID "GREEN" and this time after trying ..6Times or more  via UI.. it work!? perfect ! after all is what i want , now see if it solve the loosing connection MQTT after some day , i tell you

matjack1, 2022-03-18:
hey @pol-henri not sure if it's still interesting, but sometimes this can happen if you have the pins of the ESP connected to something while flashing. Try running the procedure again on the ESP alone, only connected to the PC via USB.

I'm closing this for now, but feel free to reopen if necessary.

closed 2022-03-18 by matjack1