omersiar/esp-rfid #407: MQTT commands List

peteshoard, 2020-10-07:
Please could someone point me at a guide to the MQTT commands that can be executed against the 1.3.3 firmware? I happily receive messages but dont know how push commands.

Thanks

ingeninge, 2020-10-08:
To send a message to your board you must form it as a JSON string. Also the IP address of the board must be included in the string. As an example here I'll show how to open the door via mqtt command. I send my messages with mosquitto_pub on linux. 
-h is the hostname or IP of the MQTT broker -t topic -m Message to be send. 
`mosquitto_pub -h Your.MQTT.Broker.IP -t rfid -m '{"doorip":"192.168.2.23","cmd":"opendoor"}'`

Commands known by developer version can be found in mqtt.esp:
```
listusr - List users
getuser - getuser
adduser - Add Users
deletusers - Delete all users
opendoor - Door open
```

peteshoard, 2020-10-08:
Thanks very much. I managed to get some idea of this from the code, but I couldn't figure out how to add a user with a type and an expiry.

The other thing that took some time to work out was what topic it subscribes too, it seems to be the root topic you add in the MQTT settings.

Appreciate your help.

pol-henri, 2020-10-08:
So whith mqttfx if whe publish on topic "rfid" with the string "opendoor", the door should open  right?

peteshoard, 2020-10-08:
Not exactly....

You must publish to the root topic that the board publishes to, this is in the 'topic' setting under 'MQTT Settings".

So if the topic is set to 'tele\esprfid' for example, you will find the results push onto your MQTT queue under the topic 'tele\esprfid\send'

You need to send 2 key value pairs as the message on the root topic as per ingeninge's comment:
`Mosquitto_pub -h Your.MQTT.Broker.IP -t tele\esprfid -m '{"doorip":"192.168.2.23","cmd":"opendoor"}'`

If you want to do other things then replace "opendoor" with the commands from above, for adding a user you need to include the name and the uid.

pol-henri, 2020-10-09:
Thanks a lot I will try this 👍

pol-henri, 2020-11-23:
i don't know how to fomat message to topic from mqttfx ,
i don't understand

for exemple if i have rfid ip :192.168.2.73 and topic set to "rfidPortail" (without "")

how format message to send opendoor ? using simply mqttfx, can u help me ? thanks a lot

thehijacker, 2020-12-02:
Use MQTT Explorer. Much easier to test mqtt commands.

Download it from here: http://mqtt-explorer.com/

[image]

I used topic "rfid" in GUI. So I send to this topic json type string:

{"doorip":"10.0.107.50","cmd":"opendoor"}

Where "10.0.107.50" is IP of the ESP-RFID board and command (cmd) is "opendoor". When I click Publish I can hear the relay on board clicking.

Firmware version 1.3.3. Still waiting for my Wiegand RFID receiver, but board is working fine for me. Powering it via 12V DC. Take note. MUST BE DC!!!  Bell transformers are not OK. They are AC and will fry the board.

I will have this  board in main electrical fuse box where my fuses are. So I ordered a DIN rail power transformator. Long looked for correct one. Finally found perfect ones from MeanWell. Model MDR-20-12 is perfect to power ESP-RFID board and 12V control current for the door lock . Also have MDR-20-24 to power the door lock and two MDR-20-5 to power two wall mounted Android tables that I will use instead of dumb two wire monitors that I currently use.

I hope I helped someone with this information.

pol-henri, 2020-12-04:
thanks for your help , unfortunately when i try to send the payload to topic it won't work for me , but the test button on UI yes ? is not the same function ? is from websocket maebe , also i see there is a list of arguments for topic like " /topic/send " but do the job for me ; thanks a lot for your detaileds explanations

thehijacker, 2020-12-04:
Today I played a bit with my Home Assistant automations. I will use ESP-RFID web GUI only for managing cards. Even that I will maybe some day bring to Home Assistant and I can open the door lock via mqtt. For me it works fine. Here are my current automations:

```
# variable to save heartbeat date and time
input_datetime:
  esp_rfid_heartbeat:
    name: ESP RFID Heartbeat
    has_date: true
    has_time: true

# Save date/time of last EPS-ID board heart beat
automation:
  - alias: "ESP RFID Save Last Heartbeat"
    trigger:
      platform: mqtt
      topic: rfid/sync
    condition:
      condition: template
      value_template: "{{ trigger.payload_json['type'] == 'heartbeat' }}"
    action:
      service: input_datetime.set_datetime
      data:
        entity_id: input_datetime.esp_rfid_heartbeat
        timestamp: "{{ trigger.payload_json.time }}"
  # Known card, notify me by Telegram
  - alias: "ESP RFID Card Scanned Known"
    trigger:
      platform: mqtt
      topic: rfid/send
    condition:
      condition: template
      value_template: "{{ trigger.payload_json['type'] == 'access' and trigger.payload_json['isKnown'] == 'true' and trigger.payload_json['access'] == 'Always' }}"
    action:
      - service: notify.telegram_andrej
        data:
          message: 'Na vhodu skenirana nova veljavna kartica: {{ trigger.payload_json.username }} ({{ trigger.payload_json.uid }})'
  # Unknow or expired card, notify me by Telegram
  - alias: "ESP RFID Card Scanned Unknown"
    trigger:
      platform: mqtt
      topic: rfid/send
    condition:
      condition: template
      value_template: "{{ trigger.payload_json['type'] == 'access' and (trigger.payload_json['isKnown'] == 'false' or trigger.payload_json['access'] == 'Denied') }}"
    action:
      - service: notify.telegram_andrej
        data:
          message: 'Na vhodu skenirana neznana ali neveljavna kartica: {{ trigger.payload_json.isKnown }} - {{ trigger.payload_json.access }} - {{ trigger.payload_json.user }} ({{ trigger.payload_json.uid }})'
  # User a Zigbee switch to open the door lock (relay on board)
  - alias: 'Switch Sonoff Open Door'
    trigger:
      platform: mqtt
      topic: "zigbee2mqtt/Switch Sonoff"
    condition:
      - condition: template
        value_template: "{{ 'single' == trigger.payload_json.action }}"
    action:
      service: mqtt.publish
      data:
        topic: rfid
        payload: "{\"doorip\":\"10.0.107.50\",\"cmd\":\"opendoor\"}"
```
You can see what mqtt events I listen and you can see what I send to open the door.

pol-henri, 2020-12-04:
yes is very clear , but don't work for me .. i have another esp-rfid and this time i try to update firmware , the older is updated via UI , but i thing is not perfectly updated (the UI show 1.04)

roysbike, 2020-12-08:
How to add user over mqtt?
`mosquitto_pub -h mqtt.360on.ru -u mqtt -P Aej9eeNgei0e -t rfid-blackcat/accesslist -m '{"doorip":"192.168.1.2","cmd":"adduser","uid":"1234","user":"test","acctype":"1","acctype2":"null","acctype3":"null","acctype4":"null","validuntil":"2145916800"}'`

windy54, 2021-02-28:
Has anyone got an example of a working python script to say open the door using MQTT.
I have got a raspberry pi interfacing to the rfid system and it also acts as a broker.
If I use the command line to send a message via mosquito pub as above the door opens.
When I create a python script to send the message it does not work. 
the message looks identical to what I send via the command line, obviously there is something wrong!