omersiar/esp-rfid #410: RDM6300 requirements

KillerDogs, 2020-10-25:
Good evening friends, I'm trying to use the project with the RDM6300, the ESP8266 module and a simple relay, but I don't know if I'm missing the pins or anything related to the voltage, I would like to know if I need a 3v to 5v converter  (I think the relay and rdm6300 need 5v), please, someone who has already used this module could show me how the scheme is used, thank you very much!

dbvcode, 2020-11-03:
@KillerDogs RDM6300 takes in 5V. Don't know what it outputs. I just connect it to the RX pin on the board, selected it in the esp-rfid menu and it works with no level shifter or anything.
Some people suggest you use a 1k-5k resistor between the reader's TX and the esp8266's RX to prevent excess current from going into the board. I did not use it.
Don't know about the relay though, I have some that need 5V and others 3.3V to trigger them.

KillerDogs, 2020-11-03:
@dbvcode 

please see if it’s to work this way I’m testing
[image]

[image]

[image]

 To test I'm going to the users tab and trying to add a new one by touching the card to the reader, is that correct?

dbvcode, 2020-11-03:
Settings seem ok in the esp-rfid but the TX of the Reader shoul go to the RX of the ESP8266

KillerDogs, 2020-11-03:
@dbvcode I carried out the exchange of the rx for the tx but without success, I don't know why it doesn't work, I tried to change the firmware versions, but it didn't work. Do you know any other project like this, so I can do a test? thanks for your help :/

dbvcode, 2020-11-03:
Tasmota [https://github.com/arendst/tasmota/](https://github.com/arendst/tasmota/)

dbvcode, 2020-11-03:
> @dbvcode I carried out the exchange of the rx for the tx but without success, I don't know why it doesn't work, I tried to change the firmware versions, but it didn't work. Do you know any other project like this, so I can do a test? thanks for your help :/

You also

> @dbvcode
> 
> please see if it’s to work this way I’m testing
> [image]
> 
> [image]
> 
> [image]
> 
> To test I'm going to the users tab and trying to add a new one by touching the card to the reader, is that correct?

You also need to share the ground between all the devices.

KillerDogs, 2020-11-04:
Hi friend, i would like to let you know that it worked, i am very new to this and did not know that i needed to share the ground. 
Now i would like to know if your ghost card problem happens here too? does it show in the log when it detects a ghost card? and the door opens?
I will be testing this week and I will try to see if it happens to my

dbvcode, 2020-11-04:
> Hi friend, i would like to let you know that it worked, i am very new to this and did not know that i needed to share the ground.

I could tell that. :) No worries, you have to start somewhere.
> Now i would like to know if your ghost card problem happens here too? does it show in the log when it detects a ghost card? and the door opens?

Yes, my fake readings show up here as well unfortunately.
> I will be testing this week and I will try to see if it happens to my

I'm very curious if it happens to you as well.

closed 2020-11-05 by KillerDogs