omersiar/esp-rfid #411: RDM6300 fake cards showing up

dbvcode, 2020-10-29:
Hi guys,
I would like to know if anyone has encountered issues with the RDM6300 popping up fake cards, mostly 000000000000 but others like 000000078F88, 00F00000F000, F000060007F1, B59A99E5A9FA, 4FFB388FFFFC, DA9386B5EA90, DAEF6FEE2B9F show up.
I don't use this project, but I have a RDM6300 and ESP8266 setup with Tasmota. I have used the cheap RDM6300 readers and SeeedStudio Grove readers and I have the same issues.
I am curious if anyone has these kind of issue here?
My original Tasmota Issue is here, [https://github.com/arendst/Tasmota/issues/9522](https://github.com/arendst/Tasmota/issues/9522)

If anyone has any ideas, I'd appreciate them a lot.
Thanks

KillerDogs, 2020-11-03:
OFF: Good morning friend, can you help me with the scheme you used to connect rdm6300 in esp? Then I can even see if the problem happens with me!

my issue: https://github.com/esprfid/esp-rfid/issues/410

Thank  you a lot

KillerDogs, 2020-11-05:
Hello, I'm testing mine after what we talked about, and so far everything is ok, just show the cards that I put, you tried to test this system in another place, as it is a frequency sensor, something can cause interference (I think)

dbvcode, 2020-11-06:
Seems it depends on the chip/batch/luck/etc..

closed 2020-11-06 by dbvcode