omersiar/esp-rfid #412: How to disable authentication?

nickbardom, 2020-10-31:
I am trying compile esp-rfid to make it compatible to Safari browser (IOS). I am trying to find the part of the code I need to remove to disable the authentication, so it would work on Safari. Could anyone help me with that? 
As I will be using esp-rfid as an access point and I will be the only one with the password to log the access point, I don't need the authentication.

ingeninge, 2020-10-31:
I found https://github.com/esprfid/esp-rfid/blob/stable/src/websrc/js/esprfid.js
line 1359. Function login. You can try to always let it return a successful login.

Please come back and report/close the ticket.

nickbardom, 2020-10-31:
Hello,
Thanks for replying. I already tried that before, but it does not work with the ESP, only with the ws emulator. Another suggestion?

nickbardom, 2020-12-05:
Hello, any news about that?

matjack1, 2022-03-18:
hey @nickbardom I'm going to close this in favour of https://github.com/esprfid/esp-rfid/issues/302 that has more details. Please be in touch if you are still interested!

closed 2022-03-18 by matjack1