omersiar/esp-rfid #413: Cannot connect to Web-Interface

henfri, 2020-11-01:
Hello,

I was able to connect to the web-interface after flashing once (AP Mode).
But after setting up my WiFi, I do not get a reply -even to a ping- anymore:
The Serial Console shows:

> [ INFO ] ESP RFID v1.3.1
> Flash real id:   00164020
> Flash real size: 4194304
> 
> Flash ide  size: 4194304
> Flash ide speed: 40000000
> Flash ide mode:  DIO
> Flash Chip configuration ok.
> 
> [ INFO ] Config file found
> {
>   "command": "configfile",
>   "network": {
>     "bssid": "86:0D:8E:8C:F3:CE",
>     "ssid": "MyNet",
>     "wmode": 0,
>     "hide": 0,
>     "pswd": "y",
>     "offtime": 0,
>     "dhcp": 0,
>     "ip": "192.168.177.48",
>     "subnet": "255.255.255.0",
>     "gateway": "192.168.177.1",
>     "dns": "192.168.177.1",
>     "apip": "192.168.4.1",
>     "apsubnet": "255.255.255.0"
>   },
>   "hardware": {
>     "readertype": 1,
>     "wgd0pin": 4,
>     "wgd1pin": 5,
>     "sspin": 0,
>     "rfidgain": 32,
>     "wifipin": 255,
>     "rtype": 1,
>     "ltype": 0,
>     "rpin": 4,
>     "rtime": 400,
>     "buttonpin": 255,
>     "doorstatpin": 255
>   },
>   "general": {
>     "hostnm": "esp-rfid",
>     "restart": 0,
>     "pswd": "admin"
>   },
>   "mqtt": {
>     "enabled": 0,
>     "host": "",
>     "port": 1883,
>     "topic": "",
>     "user": "",
>     "pswd": "",
>     "syncrate": 180,
>     "mqttlog": 0
>   },
>   "ntp": {
>     "server": "pool.ntp.org",
>     "interval": 30,
>     "timezone": 0
>   }
> }
> [ INFO ] Trying to setup RFID Hardware
> [ INFO ] Trying to connect WiFi: MyNet
> 
> [ INFO ] WiFi STA Connected
> [ INFO ] Client IP address: 192.168.177.48
> [ INFO ] Trying to setup NTP Server
> [ INFO ] Configuration done.

It is not that the board hangs... When pushing a Button on the ESP-Boar, I get:
```
Button has been pressed
mili : 159215
activating relay now
159615
159215
400
0
deactivate relay after this
mili : 159615
```

Any Idea, what could be wrong?

Regards,
Hendrik

henfri, 2020-11-01:
Hello,

I have this issue on two different boards. (Wemos and NodeMCU)
It cannot be the boards - must be me then...

Regards,
Hendrik

ingeninge, 2020-11-02:
Did you try to do a erase_flash before flashing esprfid?

alexandrechoske, 2020-12-30:
same problem.

after doing the configs cant connect to the web interface, tried all addresses.

vetruk, 2021-03-11:
Set your Wifi password with  at least 8 characters  and should be fine.

BobbyPann, 2021-08-16:
I have experienced the same issue though I eraesed flash before flashing esp-rfid 1.3.7. and though may password is longer than 8 characters. 
Any ideas? Anyone who fixed this?

matjack1, 2022-03-18:
hey @henfri not sure if this is still relevant to you, but please try with the `dev` branch in debug mode and let me know if you have any luck.

I'm going soon to release a new version of esp-rfid with a better web files handling which hopefully will be more reliable, I've had similar problems to what you are reporting.

Aleks130699, 2022-04-11:
When the Home Assistant integration is turned on, the module stops working and the web page stops opening

closed 2022-12-04 by matjack1