omersiar/esp-rfid #427: Ethernet support? ESP32 port?

DjordjeMandic, 2021-01-16:
Did someone port this to esp32 with ethernet? Is it possible to add ethernet to this project? It would suit it more, poe + ethernet perfect combination. Wifi comm. can be easily jammed.

evazzoler, 2021-03-19:
I agree! And ethernet is more stable, almost a must when you use MQTT in a full 2.4 GHz band like when you are in a building in the city center...

evazzoler, 2021-04-05:
Potesse, add the possibilty to disable wireless and connect a cheap W5100 interface to the board.

sinclairfr, 2021-08-04:
I support this request. Ethernet would be so much more stable. Also, do you guys know how hard (or how easy) it would be to port this project to a raspberry pi ?  Thanks

mkrasselt1, 2022-03-13:
I am currently in the works to port this to ttgo esp32-poe board  and build a custom pcb head for it

https://github.com/mkrasselt1/esp-rfid

matjack1, 2022-03-13:
That's great @mkrasselt1! Are you planning to keep also the ESP8266 compatibility? If you would like to contribute back to this repo I'm happy to review your code.

mkrasselt1, 2022-03-14:
> Are you planning to keep also the ESP8266 compatibility?
If by coincidence I own the right Chip to test it against, I could do it but it would be a different software.  Which chip are you planing to use?

matjack1, 2022-03-18:
@mkrasselt1 I have a NodeMCU and a Wemos D1 at hand, but I think they are all very very similar.

mkrasselt1, 2022-03-19:
No, I mean which Ethernet module are you planning to use?

matjack1, 2022-03-19:
Ah, no I have no idea about ethernet modules yet :) If you have recommendations that would be interesting

t0bse, 2022-08-30:
any news on this?

DjordjeMandic, 2022-09-03:
> Ah, no I have no idea about ethernet modules yet :) If you have recommendations that would be interesting

You can use either the SPI ones like W5500 W5100 ENC28J60 with ESP8266 or utilize hardware eth mac of ESP32 and just add eth phy module like LAN8720. 

Keep in mind that LAN8720 is just physical layer that can be used with ESP32's intergrated ethernet mac.

For ESP8266 you must use SPI modules like W5500 W5100 ENC28J60

mkrasselt1, 2022-09-04:
So for all of you, yes I am still working on it - but I plan to update the PCB as well :))

DjordjeMandic, 2022-12-11:
Any progress by anyone?

M4rkoV2, 2023-08-10:
@DjordjeMandic did you manage to do it somehow?

DjordjeMandic, 2023-08-11:
Nope, didnt even try yet, i was looking into other solutions with home assistant. Might do it one day but i dont have ESP32 eth dev enviroment (ethernet hardware)

mkrasselt1, 2023-09-26:
for now, i am using esphome to achive my goals - so far i wont be of much help to this project :/ cant say whenever i will have time reason to look into this as well