omersiar/esp-rfid #428: Reader Type Greyed out

meijeringeu, 2021-01-17:
Hi, I'am new to this, bud need some help with setting up a Wemos D1 Mini. 
I've connected an Wemos D1 Mini (V3) and attached a RFID-RC522 
Connected as follow:
(RC522 - Wemos)
SDA - D4
SCK - D5
MOSI - D7
MISO - D6 
IRQ - Not connected
GND-GND
RST - D3
3v3 - 3v3

I flached esp-rfid (v1.3.3 download, but in UI it's seen as v 1.3.1) to my Wemos, by the .bat (option 2, then COM4)

In my network, i can access the Wemos.

When I go to "Setting" - "Hardware settings";
I can't change "Reader Type", because it's greyed out.
I also can't change the "Pin-D0" and "Pin D1"
I can change "WIFI LED", "Door status", etc. (only "Relay Pin" I also can't change).

Like U understand, I want to chanch the Reader Type because I've got a RC522.

I already reset factory default and also try to install v1.3.1 from GitHub, but without result.

Can anybody help me?

lukaszwazny, 2021-01-17:
try to flash with debug.bin

matjack1, 2022-03-18:
hey @meijeringeu I'm not sure if it's still interesting to you, but I cannot reproduce anymore. Can you please try starting from the `dev` branch? Let me know if that is still happening and I can help you out!

peterveiga, 2023-01-31:
I'm having a similar problem as above.   When I flashed 1.3.1 onto a ESP8266 for testing purposes ok but when I try to change the Reader type or the Pin D0, D1 the drop downs do not drop down and give me any other options. Any help would be greatly appreciated!  Also, I see the comment above "try to flash with debug.bin" if that is the solution could you please give a step by step on how to do that or any other solution?  Thanks!!!!!!!!!

matjack1, 2023-01-31:
@peterveiga sure, you can download the binaries at the bottom of this page: https://github.com/esprfid/esp-rfid/actions/runs/3877383061

In there you can find the debug.bin file that you can use to upload from the web UI. Let me know if that doesn't work :)

matjack1, 2023-02-02:
Great! Thank you for the feedback, I'm going to close this now then ;)

closed 2023-02-02 by matjack1