omersiar/esp-rfid #429: RC522 ，UID “0” maybe ignored

kkstun, 2021-02-08:
rfid.esp 
`uid += String(mfrc522.uid.uidByte[i], HEX);`

if uid is 1bec07ec , will be 1bec7ec . The "0x07" ->7

`		for (int i = 0; i < mfrc522.uid.size; ++i)
		{
			if(mfrc522.uid.uidByte[i] < 16) uid += "0";
			uid += String(mfrc522.uid.uidByte[i], HEX);
		}
`
work well

any other good idea?

omersiar, 2021-02-10:
Good finding, maybe we should get rid of string conversion.

windy54, 2023-01-19:
I had not seen this issue and have added the same fix to the issue I rasied 466.
They could both be combined and go with this fix or the one Hinkma suggested as a reply to 466
uid += String(mfrc522.uid.uidByte[mfrc522.uid.size - (i + 1)] < 0x10 ? "0" : "");
uid += String(mfrc522.uid.uidByte[mfrc522.uid.size - (i + 1)], HEX);

kkstun, 2023-02-28:
> 

nice

closed 2023-02-28 by kkstun