omersiar/esp-rfid #431: Where to find "generic.bin"?

BobbyPann, 2021-02-12:
Hey everyone,

i tried to flash the code onto a NodeMCU. When I executed the flash.bat and chose option 1 nothing happend. So i looked into the flash.bat and found it is supposed to upload "generic.bin" via esptool. 

The Problem: I can't find a generic.bin in the entire ZIP-folder. 
Any ideas what i'm missing?

Thanks in advance.

ingeninge, 2021-02-16:
You seem to have downloaded the source code. Please use the releases. See https://github.com/esprfid/esp-rfid#software
If it is the first time for you to flash esp-rfid, then please use the latest release of version 1 : https://github.com/esprfid/esp-rfid/releases/tag/1.0.2

BobbyPann, 2021-02-16:
Thanks a lot, it worked.

closed 2021-02-16 by BobbyPann