omersiar/esp-rfid #438: Wiegand problem

JacquesDK, 2021-04-08:
I am having a problem with my wiegand reader, every time i connect it is spamming me with random key-card codes. The keypad is not making a beep before i put a key-card in front of it but in the mean it will not stop showing random codes.

I tried different gpio pins on the nodemcu and i tried older generic. bin version.

Can anybody help?

Have a nice day and great project by the way.

Jacques

TomDrifter, 2021-04-09:
Connect the negative of the reader to negative of the board

JacquesDK, 2021-04-09:
Thanks Tom that did the trick. Have a nice weekend :-)

TomDrifter, 2021-04-09:
No problem mate. it took me a long while to figuire this out also 🙂 have a look at my PI project here: https://www.youtube.com/channel/UCYD6AYhIDXQ-IrO5CRJALjQ
[https://yt3.ggpht.com/ytc/AAUvwnh7EsQsfjY3K5J3D7jWiyChQNa6vMp9owHrMv5b=s900-c-k-c0x00ffffff-no-rj]<https://www.youtube.com/channel/UCYD6AYhIDXQ-IrO5CRJALjQ>
Furniss Theory<https://www.youtube.com/channel/UCYD6AYhIDXQ-IrO5CRJALjQ>
Videos of travelling, Technology, Hobbies and General life
www.youtube.com

Many Thanks, Tom Furniss
[cid:f861dc78-1a2f-457d-b305-a69537ef2aea]

CCTV, Security, Intercom, Access Control, Turnstile & Gate Automation Systems
01933 44 18 68 - Tom@MacSystemsltd.co.uk
www.macsystemsltd.co.uk<http://www.macsystemsltd.co.uk>

closed 2022-03-09 by matjack1