omersiar/esp-rfid #439: Serious security breach

evazzoler, 2021-04-13:
Related to issue #437 I think new issue is needed becasue it no loger is a fault but a security issue.

I finally realized that this is a security lack. Unauthorized people will have access!

1) AP normally boots in longer times than esp-rfid
2) If I have access to the main current meter (normally installed ouside a building) I turn off them on the mains.
3) esp-rfid will go to recovery mode because it can't find the AP at boot
4) I connect to the esp open network from outside
5) I test the relay and open door

Recovery mode must be (optionally) disabled and one time enabled via UART once needed!
Please, consider this security improvement!

_Originally posted by @evazzoler in https://github.com/esprfid/esp-rfid/issues/437#issuecomment-818533735_

Empire99, 2021-08-11:
You should change the admin password. It is not a security lack if you put a custom password.

evazzoler, 2021-08-12:
It seems that in recovey mode the password is the default password.
Anyway what happens of in recovey mode an "open" message is sent via mqtt?

matjack1, 2022-04-04:
@evazzoler this is now fixed with this PR: https://github.com/esprfid/esp-rfid/pull/495 in the dev branch. Thank you for reporting!

I'm closing this for now, but please reopen if necessary.

closed 2022-04-04 by matjack1