omersiar/esp-rfid #441: Multiple readers and doors

onlinegill, 2021-05-01:
Hi, I would like to use it for multiple readers and doors also having separate door access restrictions. Is there any easy way to expend it with another or custom module?

TomDrifter, 2021-05-03:
i have done this on a raspberry pi. it has a custom web interface to add users etc. i built it in a enclosure with a touchscreen but had to use rfid addressing readers so the software knew which reader was what door etc.

Many Thanks, Tom Furniss
[cid:608b654e-ded4-49a6-b3f6-f55efa5dd587]

CCTV, Security, Intercom, Access Control, Turnstile & Gate Automation Systems
01933 44 18 68 - Tom@MacSystemsltd.co.uk
www.macsystemsltd.co.uk<http://www.macsystemsltd.co.uk>

onlinegill, 2021-05-03:
> i have done this on a raspberry pi. it has a custom web interface to add users etc. i built it in a enclosure with a touchscreen but had to use rfid addressing readers so the software knew which [...]

Tom, 

Could you please share script ?

giutrec, 2022-01-22:
@onlinegill I think this can help you, its support the version with single relay board, but you can edit it ;)
https://github.com/marelab/rfid-door-node-red
https://flows.nodered.org/flow/b9860e96b4df2c1a94a864b6343b067b/in/9RgOMM3J9oyX

rktaiwala, 2022-11-17:
> 

Could you please share more details?

rktaiwala, 2022-11-17:
> i have done this on a raspberry pi. it has a custom web interface to add users etc. i built it in a enclosure with a touchscreen but had to use rfid addressing readers so the software knew which [...]

Could you please share more details?