omersiar/esp-rfid #442: Crash on log view

cortexgaming101, 2021-05-03:
Whenever I leave the ESP running for a few days and then try to view the log file, it reboots. The only way to view the file is to delete it and then it starts working again for a while.

sinclairfr, 2021-11-02:
I have the exact same issue and my guess is the ESP8266 is not powerful enough or hasn't enough memory to achive this. That's why I'm aiming at porting this project to ESP32.
I will try my best.

matjack1, 2022-03-09:
@cortexgaming101 @sinclairfr I'm working on something to fix this, I'm pretty sure it's not a memory issue, but more likely a memory leak or an implementation bug that causes the reboot. I was monitoring the available memory and it was never running out, there might be either a bug or simply a slow process that doesn't let the main loop run, which triggers the watch dog timer.

Have you run the debug mode attached to the serial port? Can you maybe report what happens just before the restart?

Anyway I think I'll provide a fix soon for this! :)

mkshb, 2022-04-12:
@matjack1 
The log view problem is still there, right?

Here is my debug log when i open up user logs:

`[ INFO ] Mqtt Publish:{"type":"heartbeat","time":1649806520,"uptime":47,"ip":"10.10.10.238","hostname":"esp-rfid"}
[ INFO ] Nextbeat=1649806520
[ INFO ] Mqtt Publish:{"type":"heartbeat","time":1649806521,"uptime":48,"ip":"10.10.10.238","hostname":"esp-rfid"}
[ INFO ] Nextbeat=1649806521

Exception (3):
epc1=0x401011a5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40035650 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffe8c0 end: 3fffffb0 offset: 01a0
3fffea60:  00002678 000004cf 000004cf 401012a8
3fffea70:  00000004 3fff5144 00000004 00000001
3fffea80:  00000000 0000036c 00000020 40101533
3fffea90:  40510000 3fff5df4 00000000 00000001
3fffeaa0:  00000000 0000036c 3fff5144 4020c5ff
3fffeab0:  00000000 00003836 00000044 00000000
3fffeac0:  3fff52bc 00000001 00000001 3fff564c
3fffead0:  00000004 00000000 3fffeb50 402083da
3fffeae0:  00001968 00003836 00000000 0000036c
3fffeaf0:  3fff5144 0000036c 3fff493c 4020c7ab
3fffeb00:  3fff1344 4020d2b8 00000000 0000036c
3fffeb10:  3fff567c 3fff493c 3fff5654 4020d1a5
3fffeb20:  00000000 3fff493c 3fff5654 4020d236
3fffeb30:  3fff1344 3fff5654 3fff493c 4020d2ca
3fffeb40:  3fff50cc 00000000 3fff52bc 4020873c
3fffeb50:  3fff5629 3fff50cc 3fff5124 4020d302
3fffeb60:  3fff50cc 00000000 3fff1344 4026141e
3fffeb70:  3fffecc0 00000001 3fff56d4 3fff57a0
3fffeb80:  3fff5df4 3fff5ed4 00000000 40219ede
3fffeb90:  3fff5ec4 3fff5df4 3fffeb20 40204620
3fffeba0:  3fff5e04 3fffeb78 80003233 74616c2f
3fffebb0:  00000000 00000000 00000000 00000000
3fffebc0:  3fff1be4 00000000 40225d10 00000000
3fffebd0:  000003e8 0000bff8 00000000 00000000
3fffebe0:  00000000 00000000 3fff1be4 00000000
3fffebf0:  3fff5200 0043004f 80ffec30 3fff5e04
3fffec00:  3fffeb78 3fff5ea4 3fffeb78 4021a234
3fffec10:  3fffecd0 00000000 00000020 3fffeca0
3fffec20:  00000004 00000003 3fffecd0 40261533
3fffec30:  3fff4bd4 000f000f 00ffec73 4021a04e
3fffec40:  00000000 3fff5654 00000003 40262678
3fffec50:  6d6d6f63 00646e61 6c746567 73657461
3fffec60:  676f6c74 67617000 69660065 616e656c
3fffec70:  2f00656d 6574616c 6f6c7473 736a2e67
3fffec80:  65006e6f 6f6c7473 736a2e67 7d226e6f
3fffec90:  00000000 3fff5654 00000040 40262411
3fffeca0:  3ffeb635 401053db 3fff688c 3fff688c
3fffecb0:  3fff885c 3fff888c 00000000 3fff2b08
3fffecc0:  3fff887c 3fff885c 3ffefb20 40102ea0
3fffecd0:  3fff5634 000f000f 00000000 3fff2b08
3fffece0:  00007fff 02ed40b8 40103362 00000100
3fffecf0:  3ffebebc 7fffffff 00002200 00000001
3fffed00:  00000001 00004288 3fff56d4 40244e92
3fffed10:  00001980 00000330 00000330 401012a8
3fffed20:  3ffebee0 2c9f0300 4000050c 3fff4bd4
3fffed30:  00001a00 00000380 00000380 401012a8
3fffed40:  00000000 00000000 3fffec50 00000000
3fffed50:  3fffeca0 00000040 00000020 4010156a
3fffed60:  3fff4a1a 00000002 00000020 00000000
3fffed70:  3fff5694 00000040 3fff5654 40262a52
3fffed80:  3fff5278 00000000 00000000 401012a8
3fffed90:  00000000 4bc6a7f0 0000bfd1 3fff5654
3fffeda0:  3fff5278 3fff5654 00000004 4022420a
[... 64 lines cut ...]
3ffff1b0:  feefeffe feefeffe feefeffe feefeffe
3ffff1c0:  feefeffe feefeffe feefeffe feefeffe
3ffff1d0:  feefeffe feefeffe feefeffe feefeffe
3ffff1e0:  feefeffe feefeffe feefeffe feefeffe
3ffff1f0:  feefeffe feefeffe feefeffe feefeffe
3ffff200:  feefeffe feefeffe feefeffe feefeffe
3ffff210:  feefeffe feefeffe feefeffe feefeffe
3ffff220:  feefeffe feefeffe feefeffe feefeffe
3ffff230:  feefeffe feefeffe feefeffe feefeffe
3ffff240:  feefeffe feefeffe feefeffe feefeffe
3ffff250:  feefeffe feefeffe feefeffe feefeffe
3ffff260:  feefeffe feefeffe feefeffe feefeffe
3ffff270:  feefeffe feefeffe feefeffe feefeffe
3ffff280:  feefeffe feefeffe feefeffe feefeffe
3ffff290:  feefeffe feefeffe feefeffe feefeffe
3ffff2a0:  feefeffe feefeffe feefeffe feefeffe
3ffff2b0:  feefeffe feefeffe feefeffe feefeffe
3ffff2c0:  feefeffe feefeffe feefeffe feefeffe
3ffff2d0:  feefeffe feefeffe feefeffe feefeffe
3ffff2e0:  feefeffe feefeffe feefeffe feefeffe
3ffff2f0:  feefeffe feefeffe feefeffe feefeffe
3ffff300:  feefeffe feefeffe feefeffe feefeffe
3ffff310:  feefeffe feefeffe feefeffe feefeffe
3H�vL4!���b��
`

matjack1, 2022-06-29:
hey @mkshb sorry for the long delay here, but if you are still interested I think now the bug is fixed, if you can reproduce again with the latest dev version let me know!

matjack1, 2023-01-22:
@cortexgaming101 @sinclairfr @mkshb can you please try again with the fixes of this PR: https://github.com/esprfid/esp-rfid/pull/577 ? 

I wasn't able to reproduce anymore! :) I'm going to close this issue now, but if it's not fixed, please let me know

closed 2023-01-22 by matjack1