omersiar/esp-rfid #443: Connect to Domoticz over MQTT

djpostka, 2021-05-15:
Hello,

I would like to connect the esp to Domoticz via MQTT, however, Domoticz works with IDX (identifiers) but the configuration of the ESP does not allow me to enter an IDX.
Do you have a solution? Thank you.

matjack1, 2022-03-09:
hey @djpostka can you please give me a bit more info on what you would like to do? What should an IDX be? Do you need an extra field on the MQTT message?

marktn, 2022-03-22:
I think he wants to send the RIFD status to Domoticz. Something i also want.

I have a MQTT server and node red. I am not a mqtt expert, but it looks that the RIFD can not send mqtt commands?
Looks like the same question, https://github.com/esprfid/esp-rfid/issues/404

djpostka, 2022-03-22:
> hey @djpostka can you please give me a bit more info on what you would like to do? What should an IDX be? Do you need an extra field on the MQTT message?

Hello,

Yeah, Domoticz user an extra fiels named "IDX" for the device identification on "Dummy device/virtual device". The domoticez docs explain this: https://www.domoticz.com/wiki/MQTT

matjack1, 2022-03-22:
thank you @djpostka, I see now. Yes, we would need to add an IDX field for each relay (at least, maybe also for the leds, buttons?) so that you can configure them in esp-rfid and use them in the MQTT messages, right?

I think there needs to be a new page in the settings where you set all the IDXs for all the things that you want to expose and then if the Domoticz integration is enabled, pass the extra IDX in all the MQTT calls.

I'm not familiar with Domoticz and I'm not interested in developing it myself, but I can offer code reviews and suggestions if someone wants to implement it.

marktn, 2022-03-24:
> thank you @djpostka, I see now. Yes, we would need to add an IDX field for each relay (at least, maybe also for the leds, buttons?) so that you can configure them in esp-rfid and use them in the [...]

Yes that would be nice! Maybe a idx for each user. The you can see in domoticz who is logging in, and use it for switching lights, warnings, alarm etc..
Idx for Status door sensor, and idx for opening a door.

matjack1, 2022-03-27:
>Is it possible to let the mqtt send json to node red?

Sure! It's already sending MQTT messages on a lot of actions, it's Node Red that should read those messages and do something accordingly.

matjack1, 2022-03-31:
Did you configure the MQTT broker correctly? Are you listening to the right MQTT topics?

You should find here: https://github.com/esprfid/esp-rfid/blob/dev/README-ADDON.md the documentation around the MQTT topics that are used, OK? Let me know how it goes!

marktn, 2022-04-02:
Hi,

I think it is correct, maybe i am missing something...
In the MQTT explorer and serial it all looks fine.

[image]

[image]

[image]

matjack1, 2022-04-05:
Hey @marktn I think we are hijacking this issue to something else from the Domoticz integration? :)

Maybe you can open a separate issue where we continue discussing about the Nodered integration. While we are on the topic though, can you help me reproduce your setup so that I can try understanding better how this works? I'm not familiar with Nodered, sorry.

marktn, 2022-04-07:
Yes and no. When MQTT to Node-red works, it can send the information to Domoticz.
I will make a seperate issue! MQTT to node-red.