omersiar/esp-rfid #447: I have connected the MFRC522 correctly but my tag isn't detected

andrea006, 2021-06-09:


matjack1, 2022-03-09:
hey @andrea006 if you are still interested can you please try using the `dev` branch and build the debug target so that we can see more output on the serial port and I can try helping you troubleshooting. Good luck!

mkrasselt1, 2022-03-12:
just to make shure you are using the right token - does your anroid phone detect the token when you hold it against the nfc antenna ? because if not, it might be a 25kHZ token and unsuaitable for this project

windy54, 2022-06-25:
I have got this issue with the dev branch and a debug build.
The hardware is okay because I rolled back to the version we use in our hackspace and that reads cards.

windy54, 2022-06-27:
I have investigated this with my MFRC522 reader and from what I can see rfidState is never set to card swiped.
mfrc522Read updates the uid and type but does not update rfidState.
So I have updated as shown below.
`#ifndef OFFICIALBOARD
void mfrc522Read()
{
	
	if (!mfrc522.PICC_IsNewCardPresent() || !mfrc522.PICC_ReadCardSerial())
	{
		return;
	}
	
	mfrc522.PICC_HaltA();
	cooldown = millis() + COOLDOWN_MILIS;
#ifdef DEBUG
	Serial.print(F("[ INFO ] nfrc522Read PICC's UID: "));
#endif
	for (int i = 0; i < mfrc522.uid.size; ++i)
	{
		uid += String(mfrc522.uid.uidByte[i], HEX);
	}
#ifdef DEBUG
	Serial.print(uid);
#endif
	MFRC522::PICC_Type piccType = mfrc522.PICC_GetType(mfrc522.uid.sak);
	type = mfrc522.PICC_GetTypeName(piccType);
	rfidState = cardSwiped;
#ifdef DEBUG
	Serial.println(" " + type);
#endif
}`

Adding the line rfidState = cardSwiped afer the card type is read.

I am still investigating as the card was read, but when I updated the card information the ESP re-set.

I will post further updates and then maybe try a pull request if I sort it.

Stevce

windy54, 2022-07-02:
For some reason my SPIFSS was write protected, I reformatted the ESP and it has been ok.
I am onto my next problem.

matjack1, 2022-10-22:
looks like a duplicate of this: https://github.com/esprfid/esp-rfid/issues/506

I'm closing this one, but feel free to reopen if necessary.

closed 2022-10-22 by matjack1