omersiar/esp-rfid #448: Cannot download user file with 129 users

windy54, 2021-06-23:
We have been using this project in our hackspace for the last two years, currently working off of the Dev branch with some local mods to fix the restart issue.

This release has been used for the last 3 months, apologies I can't give a link back to the actual version.

In the last two weeks when I come to add a new member the web interface sticks at 
"Please wait while fetching data ....  Loading 1/9"
It currently won't download any logs from SPIFFs as well.

A fortnight ago when it happened I "fixed" it by restarting several times and deleting the event and access log.

There is over 1 Meg of spare SPIFF storage.

Has anyone else had this issue?

I have scanned through all the open and closed issues and can't see it being raised before, apologies If I have missed it.

So this means I can't add any new users.

Steve

omersiar, 2021-06-24:
Thank you for reporting this.

I personally did not test esp-rfid against longer period of usage.  What you can do is basically troubleshoot the issue with debugging firmware, coding and luck.

I wanted to refactor log handling, rotation and sanity checks for a long time but could not find time to polish them.

windy54, 2021-06-29:
Just an update, I have enabled DEBUG and rebuilt the software. Captured the following output:

`
[ INFO ] received websocket command userlist
[ INFO l208] WebSocket[/ws][1] error(22581): {"command":"userlist","page":8}@⸮{"command":"userlist","page":8}
[ INFO ] received websocket command userlist
[ WARN ] WebSocket[/ws][1] error(1002): Invalid UTF-8 in text frame8}@⸮⸮m⸮mmand":"userlist","page":8}
`
l208 is the line number in the code where the error is reported, something I added.
So there are 9 pages of users and it stopped on page 8 when the Invalid UTF-8 was reported.
I think the earlier error 22581 can be ignored, I can't find it listed anywhere in websocket and also at this point everything is working.
When the Invalid UTF error is reported there is a similar error on the webpage.
I get the same problem when I use Chromium on my Raspberry PI, Chrome on my MAC and also Chrome in an icognito window.
I am getting the same error if I try to download current users, or restore a file with just 8 users.

I have found this possible solution which I will look into
https://blog.fgribreau.com/2012/05/how-to-fix-could-not-decode-text-frame.html 
but it means changing the encoding on the webpage and in the software.

Steve

matjack1, 2022-02-23:
Hey @windy54 I think I was consistently able to reproduce a bug like yours, and I think I have fixed it with https://github.com/esprfid/esp-rfid/pull/484

Maybe it's now too late, but would be nice if you wanted to have a go :)

I'm closing this for now as I think it's quite old and hopefully fixed, but if you have any feedback feel free to reopen!

closed 2022-02-23 by matjack1

windy54, 2022-04-21:
Hi Mateo,
I am ready to try this, I have downloaded the source and built it, uploaded it and it does not work.
Is there a pre-compiled binary I can use?

I have tried looking through GitHub but can’t find one.

Cheers
Steve Gale